ARMmbed/yotta #560: yotta login & Continuous Integration

markus-becker-tridonic-com, 2015-11-04:
We plan to include our yotta modules in a Continuous Integration tool. How do you configure yotta to be able to do that? ```yotta login``` will not work since that spawns a browser.

autopulated, 2015-11-05:
Login information is stored in `~/.yotta/config.json` (you can set a `YOTTA_USER_SETTINGS_DIR` environment variable to change where this file will be read from), so the best thing to do is pre-populate that with credentials for a test user.

The `--noninteractive` and `--plain` flags are also useful when running in CI: (`yotta --noninteractive --plain <subcommand>`), they will prevent coloured output, and if additional login is required for any reason (for example to access a private github repo) then yotta will exit immediately instead of blocking.

markus-becker-tridonic-com, 2015-11-06:
Could the keys/tokens be passed on the command-line to yotta as well instead file-based?

autopulated, 2015-11-06:
Yes – you can actually set anything in the settings as an environment variable also. These are the ones you'll need:

```
YOTTA_KEYS_PRIVATE="private-key-variable"
YOTTA_KEYS_PUBLIC="public-key-variable"
YOTTA_GITHUB_AUTHTOKEN="authoken"
```

The format is [`YOTTA_<sectionname>_<variablename>`](https://github.com/ARMmbed/yotta/blob/master/yotta/lib/settings.py#L154) (with names converted to uppercase).

markus-becker-tridonic-com, 2015-11-09:
This seems to work fine for Python2. With Python3 ```keys.private``` is extended to something like ```YOTTA_K_E_Y_S___P_R_I_V_A_T_E``` instead of ```YOTTA_KEYS_PRIVATE```. A fix that is working on Python2 & Python3 will be merge requested soon.

closed 2015-11-24 by markus-becker-tridonic-com

autopulated, 2015-11-25:
:thumbsup: Thanks, forgot to close this with the release of 0.9.4 which fixed it.