ARMmbed/yotta #563: there should be a keyword to specify filename patterns for build artifacts

rgrover, 2015-11-06:
applications for the nRF family of targets result in a '-combined.hex' suffix. Binary files in ELF and BIN formats are also generated as a result of builds, but they are not directly useful. Test automation tools need to know which files contain installable applications. 

CC: @simonqhughes @PrzemekWirkus

autopulated, 2015-11-06:
I'm not sure the format that is used to load onto a target is implied by the target description used (as you mention, some yotta targets generate multiple build outputs that can be consumed by different tools, including ELF files for debuggers, bin files to load directly on most mbed boards, and on some targets .hex files also): the format you need to use depends primarily on the tool used for loading.

Some yotta targets also generate different versions to be used with/without a built-in bootloader, for example: which of those you want depends on whether you are using an internal or external method to update the firmware.

As a thought experiment: how would you handle an update to the firmware for a board that changed the format it was necessary to load onto it, or the case where the recommended loading method switched to be using a bootloader capable of over-the-air update?

That said, you could certainly invent a convention to use in target keywords `"mbed-loadable-output:hex"` for example, to describe in the target description which output it expects to be loadable, even if these keywords are strongly coupled to the test infrastructure used.

If you're using `yotta test` to run tests, then the test-loading script defined by your target description could specify which output to load to any files it calls (for example, you could use `$program.hex` in the test script to reference [the .hex file here](https://github.com/ARMmbed/target-nrf51822/blob/master/nordic-nrf51822-16k-gcc/target.json#L63))

rgrover, 2015-11-09:
@PrzemekWirkus: what convention would you recommend to pass information to 'yotta test'/'greentea' about the filename used for programming a target?  Should we use `mbed-interface-load-format:-combined.hex`, or should we update the target description to specify some parameter to the program for yotta test?

PrzemekWirkus, 2015-12-07:
Yes, I was speaking with Brendan how to do this. We had an initial thought how to do it well.
Now it is inside make files (e.g. -combined.hex) and we want to make it a part of configuration :)
Please be patient.