ARMmbed/yotta #566: target.json inherits from git doesnt appear to support branches

jacobrosenthal, 2015-11-09:
Its obviously idea to store features in branches, so something like this is expected
```
    "nordic-nrf51822-16k-gcc": "jacobrosenthal/target-nrf51822#fota-dat"
```
but when I use that I get
```
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt target
redbearlab-nrf51822-gcc,* missing
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt link-target redbearlab-nrf51822-gcc
info: /Users/jacobrosenthal/Downloads/ble-examples/BLE_HeartRate/yotta_targets/redbearlab-nrf51822-gcc -> /usr/local/lib/yotta_targets/redbearlab-nrf51822-gcc -> /Users/jacobrosenthal/Downloads/redbearlab-nrf51822-gcc
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt target redbearlab-nrf51822-gcc
info: get versions for nordic-nrf51822-16k-gcc
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/site-packages/yotta/main.py", line 239, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/site-packages/yotta/target.py", line 131, in execCommand
    target, errors = c.satisfyTarget(t)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/component.py", line 580, in satisfyTarget
    update_installed = update_installed
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/target.py", line 95, in getDerivedTarget
    type = 'target'
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 352, in satisfyVersion
    return satisfyVersionByInstalling(name, version_required, working_directory, type=type)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 294, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required, _registryNamespaceForType(type))
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 145, in latestSuitableVersion
    remote_component.availableTags(),
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/github_access.py", line 249, in availableTags
    ) for t in self._getTags()
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/github_access.py", line 221, in _getTags
    self.tags = _getTags(self.repo).items()
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/github_access.py", line 69, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/github_access.py", line 99, in _getTags
    repo = g.get_repo(repo)
  File "/usr/local/lib/python2.7/site-packages/github/MainClass.py", line 200, in get_repo
    "%s%s" % (url_base, full_name_or_id)
  File "/usr/local/lib/python2.7/site-packages/github/Requester.py", line 169, in requestJsonAndCheck
    return self.__check(*self.requestJson(verb, url, parameters, headers, input, cnx))
  File "/usr/local/lib/python2.7/site-packages/github/Requester.py", line 210, in requestJson
    return self.__requestEncode(cnx, verb, url, parameters, headers, input, encode)
  File "/usr/local/lib/python2.7/site-packages/github/Requester.py", line 247, in __requestEncode
    status, responseHeaders, output = self.__requestRaw(cnx, verb, url, requestHeaders, encoded_input)
  File "/usr/local/lib/python2.7/site-packages/github/Requester.py", line 271, in __requestRaw
    requestHeaders
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1001, in request
    self._send_request(method, url, body, headers)
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1035, in _send_request
    self.endheaders(body)
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 997, in endheaders
    self._send_output(message_body)
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 850, in _send_output
    self.send(msg)
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 812, in send
    self.connect()
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1204, in connect
    HTTPConnection.connect(self)
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 793, in connect
    self.timeout, self.source_address)
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 553, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
socket.gaierror: [Errno 8] nodename nor servname provided, or not known
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ 
```

Merging my changes into master seems to work
```
    "nordic-nrf51822-16k-gcc": "jacobrosenthal/target-nrf51822"
```

jacobrosenthal, 2015-11-09:
Also looks like yt command line doesnt support github at all for targets?
```
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt target git@github.com/jacobrosenthal/target-nrf51822
error: Invalid target: "git@github.com/jacobrosenthal/target-nrf51822"
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt target git@github.com:jacobrosenthal/target-nrf51822.git
error: Invalid target: "git@github.com:jacobrosenthal/target-nrf51822.git"
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt target git@github.com:jacobrosenthal/target-nrf51822.git#fota-dat
error: Invalid target: "git@github.com:jacobrosenthal/target-nrf51822.git#fota-dat"
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt target jacobrosenthal/target-nrf51822.git#fota-dat
error: Invalid target: "jacobrosenthal/target-nrf51822.git#fota-dat"
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt target jacobrosenthal/target-nrf51822.git
error: Invalid target: "jacobrosenthal/target-nrf51822.git"
```

jacobrosenthal, 2015-11-09:
Related issue:

in my redbearlab-nrf51822-gcc target, I inherit from a fork of nordic's targets (but they have multiple within that repo)
```
    "nordic-nrf51822-16k-gcc": "jacobrosenthal/target-nrf51822"
```

Normally when I link, a `yt install` appears to be run for me, but it doesnt happen here. Manually running it doesnt fetch all the required dependencies and a `yt build` fails
```
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt link-target redbearlab-nrf51822-gcc
info: /Users/jacobrosenthal/Downloads/ble-examples/BLE_HeartRate/yotta_targets/redbearlab-nrf51822-gcc -> /usr/local/lib/yotta_targets/redbearlab-nrf51822-gcc -> /Users/jacobrosenthal/Downloads/redbearlab-nrf51822-gcc
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt target redbearlab-nrf51822-gcc
info: get versions for nordic-nrf51822-16k-gcc
info: download nordic-nrf51822-16k-gcc@v0.0.4 from GitHub jacobrosenthal/target-nrf51822
info: get versions for mbed-gcc
info: download mbed-gcc@0.1.3 from the public module registry
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt install
info: get versions for ble
info: download ble@2.0.3 from the public module registry
info: get versions for ble-nrf51822
info: download ble-nrf51822@2.0.2 from the public module registry
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ 
```

However if Isolate the target Im attempting to inherit from into its own repo
```
    "nordic-nrf51822-16k-gcc": "jacobrosenthal/nordic-nrf51822-16k-gcc"
```

It still doesnt install automatically `yt install` for me like a published targed would, but it DOES install everything and build successfully
```
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ rm -rf yotta_*
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt link-target redbearlab-nrf51822-gcc
info: /Users/jacobrosenthal/Downloads/ble-examples/BLE_HeartRate/yotta_targets/redbearlab-nrf51822-gcc -> /usr/local/lib/yotta_targets/redbearlab-nrf51822-gcc -> /Users/jacobrosenthal/Downloads/redbearlab-nrf51822-gcc
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt target redbearlab-nrf51822-gcc
info: get versions for nordic-nrf51822-16k-gcc
warning: Github repository "jacobrosenthal/nordic-nrf51822-16k-gcc" has no tagged versions, default branch will be used
info: download nordic-nrf51822-16k-gcc@ from GitHub jacobrosenthal/nordic-nrf51822-16k-gcc
info: get versions for mbed-gcc
info: download mbed-gcc@0.1.3 from the public module registry
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ yt install
info: get versions for ble
info: download ble@2.0.3 from the public module registry
info: get versions for ble-nrf51822
info: download ble-nrf51822@2.0.2 from the public module registry
info: get versions for mbed-drivers
info: download mbed-drivers@0.11.2 from the public module registry
info: get versions for mbed-hal
info: download mbed-hal@1.0.5 from the public module registry
info: get versions for cmsis-core
info: download cmsis-core@1.0.1 from the public module registry
info: get versions for ualloc
info: download ualloc@1.0.2 from the public module registry
info: get versions for minar
info: download minar@1.0.1 from the public module registry
info: get versions for core-util
info: download core-util@1.0.1 from the public module registry
info: get versions for compiler-polyfill
info: download compiler-polyfill@1.1.1 from the public module registry
info: get versions for mbed-hal-nordic
info: download mbed-hal-nordic@1.0.0 from the public module registry
info: get versions for mbed-hal-nrf51822-mcu
info: download mbed-hal-nrf51822-mcu@1.0.6 from the public module registry
info: get versions for mbed-hal-mkit
info: download mbed-hal-mkit@1.0.1 from the public module registry
info: get versions for cmsis-core-nordic
info: download cmsis-core-nordic@1.0.1 from the public module registry
info: get versions for cmsis-core-nrf51822
info: download cmsis-core-nrf51822@1.1.0 from the public module registry
info: get versions for dlmalloc
info: download dlmalloc@1.0.0 from the public module registry
info: get versions for minar-platform
info: download minar-platform@1.0.0 from the public module registry
info: get versions for minar-platform-mbed
info: download minar-platform-mbed@1.0.0 from the public module registry
error: ble-nrf51822 does not meet specification ^1.0.0 required by mbed-hal-nrf51822-mcu
Jacobs-MacBook-Air:BLE_HeartRate jacobrosenthal$ 
```

autopulated, 2015-11-10:
The limited command-line support for complex installation sources like branch is a known issue, but I think it should work OK in `inherits:` in the target.json file.

Note that in this case, the nordic targets are not one-per-repository, so there will not be a target.json file at the root of the repository, which is what yotta expects when downloading from github. So perhaps that is why you had a problem?

For now you will need to [use `yotta link-target`](http://yottadocs.mbed.com/tutorial/targets.html#testing-targets) to use your custom target version.

closed 2017-03-29 by thegecko