ARMmbed/yotta #577: Being able to work directly in yotta modules pulled from private repositories

javier-moreno-tridonic-com, 2015-11-17:
Most of the time that I need to pull in dependencies from private repositories, I need to work or do changes at the same time in the dependencies and the current module (specially at early stages of the development)

Achieving this kind of behavior via "yotta link" is feasible when working with few modules, but as the number increases it feels as work that could be avoided.

The suggestion would be to be able to specify (via module.json maybe?) some sort of "link behavior". For instance, let's say that I specify such "link behaviour" in the module "a_module" that I would like to use as a dependency in the executable "working module".

When I type:
```
working_module> yotta build 
```
Instead of cloning the module "a_module" in a temporary folder and copy over the contents to "working_module/yotta_modules/a_module" (as I understand yotta does currently), yotta could do the following:

- Clone the "a_module" into the YOTTA_PREFIX folder (if is not already there)
- Link the module to "your_working_module"

0xc0170, 2015-11-17:
> Achieving this kind of behavior via "yotta link" is feasible when working with few modules, but as the number increases it feels as work that could be avoided.

Me and other developers shared similar experience.

cc me

betzw, 2015-11-17:
Idem