ARMmbed/yotta #578: We need a way to report user environment in a bug report

bremoran, 2015-11-17:
When a user reports a bug in a yotta module, the module maintainer needs a way to see the environment.  Since this is an issue for most yotta modules, it would be helpful if yotta could generate an environment report.  There are lots of options for how this could be implemented.  Two suggestions have been brought up already:

1. Create a ```yotta bug-report``` command, which generates the environment dump.  Optionally, the dump could generate a submission to a designated repo, generate a case ID, etc.
1. yotta could always dump the environment report to the build directory.  This would allow a user to simply upload the environment dump file.

cc @autopulated @bogdanm @sam-taylor-arm @niklas-arm @adbridge @meriac @0xc0170 @rgrover

autopulated, 2015-11-17:
fwiw some analytics reporting of build failures (and which target/module combinations they are associated with) is planned.

What information would you include in the report?

rgrover, 2015-11-17:
Information which can be bundled for reference:
* module tree (including use of links)
* target tree (including use of links)
* toolchain versions
* yotta version
* python version

meriac, 2015-11-20:
I would also include the path-variable and the path of all tool-binaries used by yotta (cmake, gcc, linker etc.), optionally the user might want to include an elf file with debug-information for detailed help.