ARMmbed/yotta #579: No execution permissions

betzw, 2015-11-17:
After updating yotta on Linux, I always have to manually add execution permissions to
```/usr/local/bin/yt``` and ```/usr/local/bin/yotta```

autopulated, 2015-11-17:
Thanks for reporting – can I ask which linux distribution are you using, and what version of `pip` do you have (presuming you used `pip` to install the upgrade?)

betzw, 2015-11-17:
```
$ pip --version
pip 7.1.2 from /usr/local/lib/python2.7/dist-packages/pip-7.1.2-py2.7.egg (python 2.7)
$ uname -a
Linux agr011510 4.2.0-18-generic #22-Ubuntu SMP Fri Nov 6 18:25:50 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

autopulated, 2015-11-17:
Thanks – which ubuntu release is that?

betzw, 2015-11-17:
Ubuntu 15.10

autopulated, 2015-12-11:
This should have been fixed in the two most recent yotta releases – could you confirm?

betzw, 2015-12-11:
This morning I have upgraded to v0.11.0 but I still had to add execution permissions to `yt` & `yotta`.

autopulated, 2015-12-11:
Hmm, ok. Thanks for checking!

betzw, 2015-12-18:
Still not resolved for yotta-0.12.0 :-(
```
Collecting yotta
  Downloading yotta-0.12.0.tar.gz (107kB)
    100% |████████████████████████████████| 110kB 163kB/s 

...

Building wheels for collected packages: yotta
  Running setup.py bdist_wheel for yotta
  Stored in directory: /root/.cache/pip/wheels/68/93/c9/34cbf5bd17ba1e399cd024f9f447e392b3814b373798fcd0ca
Successfully built yotta
Installing collected packages: yotta
  Found existing installation: yotta 0.11.0
    Uninstalling yotta-0.11.0:
      Successfully uninstalled yotta-0.11.0
Successfully installed yotta-0.12.0
betzw@agr011510:~$ yt --version
bash: /usr/local/bin/yt: Permission denied
```

autopulated, 2015-12-18:
:cry: 

The scripts in the archive on PyPI definitely have the +x bit set correctly. Could you try maybe cloning the source, `chmod +x ./bin/*`, and then `pip install .` – and check if the local pip install sets the permissions correctly?

betzw, 2015-12-18:
This works.
I am just wondering which files you were referring to when talking about "The scripts in the archive on PyPI definitely have the +x bit set correctly"?

autopulated, 2015-12-18:
Hmm OK, that's really strange.

The two scripts are in the `./bin` folder of the source (https://github.com/ARMmbed/yotta/tree/master/bin) / the corresponding tarball which pip downloads when it installs yotta is here: https://pypi.python.org/packages/source/y/yotta/yotta-0.12.0.tar.gz#md5=cadf95ab334c4d824bf55af785e4edc9

betzw, 2015-12-18:
Well, on the local installation I did the scripts (i.e. `yt` & `yotta`) in `./bin` - even having the right permissions - are not exactly the same as the ones in `./usr/local/bin/` and the ones `/usr/local/bin`. While the scripts in the latter two directories are the same.
Here is the `diff` for `yotta`:
```
betzw@agr011510:~/spielwiese/mbed/yotta$ diff ./bin/yotta ./usr/local/bin/yotta 
1c1
< #! /usr/bin/env python
---
> #!/usr/bin/python
```
Any idea, why is that?

betzw, 2015-12-21:
It's getting even more weird:
Launching ` sudo -EH pip install .` from my local source copy installs the scripts with execution permissions in `/usr/local/bin/`.
Seems as if cloning 'https://github.com/ARMmbed/yotta.git` and running `sudo -EH pip install .` in the clone produces a different results from just running `sudo -EH pip install -U yotta` from somewhere.

autopulated, 2016-01-04:
The #! line is modified by pip when it installs the scripts, so that's normal.

I'm completely out of explanations though, can only suggest reporting as an issue [on pip](https://github.com/pypa/pip/issues?utf8=✓&q=is%3Aissue+is%3Aopen)!

Possibly we should just go back to using entry_points instead of custom scripts (on windows entry_points is used still anyway): there is a significant startup-time hit to that, but only in python environments with lots of modules installed, which is probably not typical for users.

betzw, 2016-01-08:
I do not know what exactly made the trick, but after a generic Ubuntu SW update and a forced re-install of pip, my latest upgrade of yotta (to version 0.12.1) didn't require me anymore to manually add the execution permissions.
Anyway, as you stated before, most likely my issue is more related to pip than to yotta and therefore I apologize having you bothered with it.
Regarding the question about entry_points vs custom scripts I prefer not to comment not being at all competent on the matter.

Ciao Wolfgang

autopulated, 2016-01-08:
OK, well it's good to have a record of the issue here in case anyone else does come across the same problem :)

I will close for now!

closed 2016-01-08 by autopulated