ARMmbed/yotta #580: Add command line option to specify extra config

bremoran, 2015-11-18:
To support code coverage, I need to be able to specify

```JSON
    "debug" : {
      "options" : {
        "coverage" : true
      }
    }
```

This may only be desirable for some builds, so always-on configuration in application config seems wrong.  Always-on configuration in target config seems wrong.

Therefore, it must be possible to specify code coverage debug builds. One way to do this is to specify either a config file, or a config parameter on the command line.

bremoran, 2015-11-18:
This is required to support https://github.com/ARMmbed/mbed-drivers/pull/119

autopulated, 2015-11-19:
Presumably the existing application config.json file doesn't work because you want to do this for a module and its tests?

bremoran, 2015-11-19:
That is correct

commit d07d4587 (2015-12-11): Respect the --config option

The value passed to --config is first interpreted as a file name; if loading
that file fails, literal JSON parsing is attempted.

Fixes #580
closed 2015-12-11 by autopulated