ARMmbed/yotta #585: c:\ytexe yt link does not install module

BlackstoneEngineering, 2015-11-19:
## Problem 
To add a local library module to a executable module it requires 3 steps
`c:\ytlib yt link`
`c:\ytexe yt link ytlib`
If you run `yt ls` at this point there is no indication that `ytlib` is linked into the project, you must either `yt install` or `yt build` to get `ytlib` added to your module.json file
`c:\ytexe yt install ytlib`
At this point you can now see `ytlib` in a `yt ls` command output.

### example
I'm a developer developing locally. I want to add simplelog to my project to help me debug and maybe tweak a few pretty print settings in simplelog for fun. So I make the yotta executable `ytexe`. I clone the simplelog repo to my local machine so they are both at my root directory. Now to add simplelog to my project I must 
`C:\simplelog yt link`
`C:\ytexe yt link simplelog`
at this point I have added simplelog calls into my code and run `yt build`, which fails, because the simplelog module has not been added to my module.json, nor does it show up in a `yt ls` command.
I must run `yt install simplelog`. 

This may be expected behaviour, and maybe we dont want to change it, but at the very least it makes for a bad user experience. 

## Solution
1) have a `yt link <absolute file path>` command that takes care of all these steps
or
2) have `c:\ytexe yt link ytlib` add the module to the module.json or otherwise make it obvious that it is added to the project, currently there is no feedback.
or
3) when a user runs `c:\ytexe yt link ytlib` give them a feedback message telling them they need to `yt install ytlib` to finish adding the module. I think this solution is the least optimum because it requires 3 steps instead of 1, but it is the miminum required for user interaction.

autopulated, 2015-11-19:
I think `yt unlink ...` shouldn't remove the module from the description file, so I think your solution (3) might be best, as this will keep corresponding behaviour with `link` and `unlink`.

BlackstoneEngineering, 2015-11-19:
so how do I remove a module from the executable module? what is the command to do this?
`yt install` is for grabbing from the registry or remote git repo, but it cannot do local file systems
`yt link` is for cludging local file system modules into the structure so `yt install` can grab them. Why not just enable `yt install` to be able to handle local file system files?

PR #586 (2015-11-24): support 'yotta unlink' to un-globally-link a module
 (and yotta unlink-target for targets); fixes #585 and #362
commit bfa1bce5 (2015-11-24): support 'yotta unlink' to un-globally-link a module (and yotta unlink-target for targets); fixes #585 and #362
closed 2015-11-24 by autopulated