ARMmbed/yotta #587: yotta outdated exits on modules not in the repository

markus-becker-tridonic-com, 2015-11-20:
If you have a mixture of git repos and yotta registry modules, yotta exits on the first non-registry module with this traceback:

```
$ yotta outdated
Fatal Exception, yotta=0.9.4
Traceback (most recent call last):
  File "C:\winapps\Python27\Scripts\yotta-script.py", line 9, in <module>
    load_entry_point('yotta==0.9.4', 'console_scripts', 'yotta')()
  File "C:\winapps\Python27\lib\site-packages\yotta\main.py", line 240, in main
    status = args.command(args, following_args)
  File "C:\winapps\Python27\lib\site-packages\yotta\outdated.py", line 37, in execCommand
    return displayOutdated(dependencies, use_colours=(not args.plain))
  File "C:\winapps\Python27\lib\site-packages\yotta\outdated.py", line 58, in displayOutdated
    latest_v = access.latestSuitableVersion(name, '*', registry='modules', quiet=True)
  File "C:\winapps\Python27\lib\site-packages\yotta\lib\access.py", line 108, in latestSuitableVersion
    vers = remote_component.availableVersions()
  File "C:\winapps\Python27\lib\site-packages\yotta\lib\registry_access.py", line 485, in availableVersions
    return _listVersions(self.namespace, self.name)
  File "C:\winapps\Python27\lib\site-packages\yotta\lib\registry_access.py", line 102, in wrapped
    return fn(*args, **kwargs)
  File "C:\winapps\Python27\lib\site-packages\yotta\lib\registry_access.py", line 251, in _listVersions
    'Check that the name is correct, and that it has been published.') % (name, namespace)
yotta.lib.access_common.Unavailable: openais-mbed-common does not exist in the modules registry. Check that the name is correct, and that it has been published.
```

Simplest expected behaviour would be to skip the git repositories and just check the registry modules.
Stretch target: check the git repositories tags or even updates on the specified branch.

autopulated, 2015-11-20:
Thanks – with #574 it should be possible to improve this.

mlnx, 2016-01-13:
@markus-becker-tridonic-com Are you still hitting this issue with the latest version of yotta (0.12.2) ?

autopulated, 2016-01-13:
The crash was fixed in #634. I left this open to track checking for updates from git repos.

markus-becker-tridonic-com, 2016-04-22:
Not crashing on git anymore. Issue is kept open to track 'checking for updates from git repos' Retitled this issue to reflect this.

markus-becker-tridonic-com, 2016-07-05:
Closing my yotta issues.

closed 2016-07-05 by markus-becker-tridonic-com