ARMmbed/yotta #588: module.json: licenses url field

ilg-ul, 2015-11-22:
I have a problem with the mandatory `licenses url` field.

The module I'm trying to build has a custom open source license file, specified, according to SPDX, as `"type": "SEE LICENSE IN LICENSE.pdf"`, but the license is not available at an URL.

Since the `url` field is mandatory, I had to add it with a phoney content, which is not ok.

From my experience, even for SPDX official licenses, this URL is not very useful, since it is always derived from the license name, for example:

```
"licenses": [
		{
			"url": "https://spdx.org/licenses/LGPL-3.0.html",
			"type": "LGPL-3.0"
		}
	],
```

This raises the question if the `url` field really needs to be mandatory?

If I remember right, the latest NLM specifications use only the `type` field. Also, multiple licenses are concatenated into a single `type` field.

My suggestion is to do the same, or at least to update the schema and make the `url` field not mandatory.

autopulated, 2015-11-23:
Thanks for reporting.

The `"license":"<SPDX-License-Expression>"` field is actually the preferred way of specifying licenses (the docs need updating to reflect this). You can use `"license":"LicenseRef-LICENSE.txt"` for example to refer to a license in a file.

ilg-ul, 2015-11-23:
I just tried a publish and the following NLM syntax was perfectly accepted:

```
	"license": "SEE LICENSE IN CMSIS_END_USER_LICENCE_AGREEMENT.pdf",
```

Your schema is:

```
        "licenseID": {
            "type":"string",
            "minLength":3,
            "maxLength":1000
        },
```

Where does this syntax with `"license":"LicenseRef-LICENSE.txt"` come from?

autopulated, 2015-11-23:
I believe `LicenseRef-` is a SPDX standard way of referring to a non-standard license. From http://spdx.org/rdf/spdx-terms-v2.0/index.html / http://spdx.org/rdf/terms#licenseId:

> * comment "A human readable short form license identifier for a license. The license ID is iether on the standard license oist or the form "LicenseRef-"[idString] where [idString] is a unique string containing letters, numbers, ".", "-" or "+"." (en)
> * term_status "stable" (en)

ilg-ul, 2015-11-23:
yeah, the definition is not particularly clear if the `[idString]` is a file name, and, more annoying, it does not allow `_`.

if this is the final syntax, I'll rename `CMSIS_END_USER_LICENCE_AGREEMENT.pdf` to `LICENSE.pdf ` (I hope ARM will not complain about this!) and update the module.json to `"license": "LicenseRef-LICENSE.pdf"`.

jlovejoy, 2015-11-24:
"License-Ref" comes from section 5 of the SPDX specification - see: http://spdx.org/sites/spdx/files/SPDX-2.0.pdf - that will have the full explanation. The link @autopulated referred to is the RDF terms (only). 

Also note, if you need to express a scenario where multiple licenses apply, then you can use the license expression syntax, as explained in Appendix IV of the SPDX spec :)

Hope that helps fill in some gaps!

ilg-ul, 2015-11-24:
thank you, @jlovejoy, I'll update to the new syntax.

closed 2015-11-24 by ilg-ul