ARMmbed/yotta #589: yotta unpublish?

ilg-ul, 2015-11-22:
Probably this is a rhetorical question, but is there any way to remove modules from the registry?

As always at the beginning, I did some tests and once I'll decide on the final naming scheme and module content, I'll probably want to hide the dead modules.

autopulated, 2015-11-23:
Thanks for noting that it's missing from the command reference, it should be there!

There's no need to remove test or experimental modules from the registry. When search results for the registry are displayed, the download count is taking into account, so that the real modules should be presented before any previous test modules.

The most significant factors when someone is choosing which module to download are that it has a good clear description, and a useful readme.md file with documentation on how to get started – so if you make sure these are also good for your "real" modules you shouldn't have any problems.

Hope that explains things!

ilg-ul, 2015-11-23:
Ok, I'll remove the keywords, so the modules will not show in my searches, and I'll also update the description to mention the modules should be ignored.

If you don't mind managing a repository with some degree of garbage, ok, just ignore this enhancement request, but I think that a mechanism to tidy the repository would still be useful.

autopulated, 2015-11-23:
We're perfectly reconciled with the fact that the registry is On The Internet, and therefore trying to curate it is likely Sisyphean – highlighting the most popular (and, eventually, the best tested and most compatibly licensed) modules in search is our strategy for surfacing the best content.

autopulated, 2015-11-23:
(leaving this open as a bug to track the lack of documentation for `yotta unpublish`)