ARMmbed/yotta #590: Public Registry acceptable use policy question

ilg-ul, 2015-11-23:
Is there any document defining the registry AUP?

In the [ARMmbed page](https://www.mbed.com/en/development/software/mbed-yotta/) it is stated:

> ... we encourage other C and C++ software projects to adopt it as well. Even outside mbed, yotta enables wider reuse of C-family software modules and accelerates application development.

I'm the maintainer of [GNU ARM Eclipse](http://gnuarmeclipse.github.io) and I'm currently working on improving the project template wizard by using a packaging system that supersedes the CMSIS Packs specifications. The new packages are all C/C++ code, some of them streamlined versions derived from existing CMSIS Packs, but with a more elaborate metadata (called [XCDL](http://xcdl.github.io), as an homage to eCos CDL).

Initially I planned to use a [private static repository](http://xpacks.github.io/index/), with direct links to packages, but I think a public repository, like yotta, is preferable.

If you want to take a look, a preliminary module, currently published as the `ilg-arm-cmsis` yotta module, is available from https://github.com/xpacks/arm-cmsis.

Can you confirm that such specialised modules are within your acceptable use policy, and I can use the yotta repository to store and retrieve these packages that will be accessed by **GNU ARM Eclipse**?

Thank you,

Liviu

autopulated, 2015-11-23:
The expectation is that modules published to the yotta registry are consumed and built by yotta, or in a way compatible with that (so that they are potentially available to people building applications for mbed). So if the modules that you publish are compatible with being used by yotta, I don't see why any problem would arise.

The ilg-arm-cmsis module, as currently packaged, won't work as a yotta module though, because the source files are not exposed to yotta (either in the `source` directory for the [automatic build system](http://yottadocs.mbed.com/reference/buildsystem.html#automatic), or via [a custom CMake](http://yottadocs.mbed.com/reference/buildsystem.html#custom-cmake) build description). The header files also assume that un-prefixed names can be used to include files in subdirectories, which is strongly advised against in yotta, due to the possibility of header name clashes across different modules.

ilg-ul, 2015-11-23:
> The expectation is that modules published to the yotta registry are consumed and built by yotta

I can probably arrange for a custom CMake, but my modules better fit the _... we encourage other C and C++ software projects to adopt it as well. Even outside mbed ..._, but even if not directly used by mbed, it is for ARM projects anyway.

autopulated, 2015-11-23:
Regardless of use on ARM or elsewhere, we're trying to create an ecosystem of re-usable software modules – which means more than just putting everything in one place, it means making sure that everything can, as much as possible, work together.

So if your modules work, that's great, regardless of the target hardware!

ilg-ul, 2015-11-23:
my xPack/XCDL modules are much more than simple libraries or executables, they are intended as components to be compiled together (not linked together) in an embedded application; they have a much richer set of metadata, intended to guide the process of creating applications from components under GUI (or script) control and the build process.

the rationals behind such an approach are documented at http://xcdl.github.io/guide/rationals/ and the main concepts are defined in http://xcdl.github.io/guide/concepts/ (work in progress).

on the down side, even if I can add custom CMake code to build generic libraries from some components, I doubt my modules would be really useful in the traditional style, since most of the components depend on configuration data, reflected in generated headers and sources, that are strictly customised to match the target application, and I don't know if this is currently supported by yotta.

ilg-ul, 2015-11-23:
> The ilg-arm-cmsis module ... won't work as a yotta module ... the header files also assume that un-prefixed names can be used to include files in subdirectories, which is strongly advised against in yotta, due to the possibility of header name clashes across different modules.

one comment, for completeness: the ilg-arm-cmsis module repacks the ARM CMSIS headers, which, like it or not, are now industry standard. I confess I don't like the way these headers are designed and used, but I'm afraid there is not much I can do to change them. perhaps you, at **ARM mbed**, can do more.

hugovincent, 2015-11-23:
Hi @ilg-ul, you're very welcome to use the yotta registry to host any yotta modules (meaning: they can be managed and built using yotta for at least one yotta target), whether they target ARM CPUs or not, whether they're in any way related to mbed or not. However the registry is not a general purpose file storage service. For storing arbitrary files (e.g. things that aren't yotta modules), maybe try Amazon S3 or Azure Blob store, or any of the many other choices.

We haven't yet published a formal acceptable use policy, but for now it probably suffices to say: we reserve the right to remove anything from the registry that is illegal or offensive, and anything that isn't a valid yotta module. By this reasoning, I would argue that http://yotta.mbed.com/#module/ilg-arm-cmsis is not a valid yotta module, since it can't be built by yotta.

Also, if you would like to provide feedback on the design of CMSIS headers, please email me at `hugo.vincent at arm.com` and I'll make sure the correct people see it.

ilg-ul, 2015-11-23:
> the registry is not a general purpose file storage service.

I'm not interested in a storage service, GitHub Releases and SourceForge Files are more than enough for my needs. 

What I'm interested is a registry service, i.e. a system where multiple people can contribute code without me having to approve additions to a public index, which I already have.

> we reserve the right to remove ... anything that isn't a valid yotta module.

As I said, I can probably tweak my modules to pass a build, but they'll be totally useless outside my environment.

So, I take your answer as a NO.

Please update the project documentation to clearly mention that the registry cannot be used for _ other C and C++ software projects_, as it says now.

> the design of CMSIS ...

The entire design is way too C specific, making usage in C++ more complicated than it should be; the CMSIS OS makes heavy use of macros to define objects, and these macros look like (very confusing) function calls; the CMSIS Drivers pass the pointer to the object as the last parameter, making very difficult to implement them in C++; the general naming convention is very inconsistent; compiling even the CORE headers trigger many conversion warnings, which might not be dangerous, but we never know.

From my point of view, a better approach would be to design a new version (CMSIS++?) and add a C interface to it.

hugovincent, 2015-11-23:
> Please update the project documentation to clearly mention that the registry cannot be used for _ other C and C++ software projects_, as it says now.

yotta is designed to be used by a wide range of C and C++ projects (irrespective of CPU architecture or OS), but does provide its own build system based around CMake, configuration management, dependency tracking, directory layout, etc. Thus, you can use it for any kind of project, provided you're willing to let yotta be responsible for those things. Similarly, the yotta registry is designed to address the needs of a registry in yotta-based projects. Internally, we've used yotta to build applications on other platforms including Linux command line tools, OS X applications etc. So, I think the documentation is correct and fair, provided you're willing to let yotta do the pieces of functionality it was designed to do.

autopulated, 2015-11-24:
Thanks @hugovincent – yes you can use yotta for any project: embedded/desktop/ARM or non-ARM, and that's what the language on the website refers to.

ilg-ul, 2015-11-24:
> On 23 Nov 2015, at 23:39, Hugo Vincent <notifications@github.com> wrote:
> 
> Please update the project documentation to clearly mention that the registry cannot be used for _ other C and C++ software projects_, as it says now.
> 
> yotta ... does provide its own build system based around CMake, configuration management, dependency tracking, directory layout, etc. Thus, you can use it for any kind of project, provided you're willing to let yotta be responsible for those things.

what if CMake does not match my needs? what if the yotta application itself, with its Python environment, does not match my Eclipse environment?

my plan was to implement part of yotta functionality as an Eclipse plug-in, part of GNU ARM Eclipse, and integrate the build into my build environment.

> Similarly, the yotta registry is designed to address the needs of a registry in yotta-based projects. Internally, we've used yotta to build applications on other platforms including Linux command [...]

what if for my project it is more appropriate to do the publish/search/install/update/etc directly from Eclipse, without yotta?

I don't doubt yotta is able to build Linux command line tools. what I doubt is that yotta is able to build reasonably complex embedded applications.

what is your solution to build modules that depend on some specific application configuration? in other words, the module includes headers that are specific to the application, even headers/source files that are generated by a tool during application configuration?

ilg-ul, 2015-11-24:
>  yes you can use yotta for any project: embedded/desktop/ARM or non-ARM, and that's what the language on the website refers to.

ok, thank you, I will, just that I'll probably not use yotta, but the Eclipse plug-in. and some of the modules will provide very limited functionality when used outside my environment.

autopulated, 2015-11-24:
yotta has a global [configuration system](http://yottadocs.mbed.com/reference/config.html) for any modules which need app-specific configuration.

It's true though that re-using software in embedded applications is much harder than in other environments, not least because almost every extant embedded software package breaks some rule of re-usability, often assuming that it is the primary component of a system around which everything else will fit.

There is no cost in using namespaces (or prefixes, if C++ name mangling is unacceptable), or in using the full path to include a header, and there is very little cost in eliminating unnecessary global state, and in making headers behave consistently wherever they are included from. These are things that make software more re-usable however it is built or distributed, and something that we all should be doing.

ilg-ul, 2015-11-24:
> It's true though that re-using software in embedded applications is much harder

indeed.

> yotta has a global configuration system for any modules

yeah, I took a look; what can I say? if it is enough for your needs, fine, use it. 

however generating preprocessor lines with a rigid naming scheme is not exactly my view of a configuration system. what if I need to generate C++ constexpr statements? or even more complex things, like structures? what if configuration options have dependencies between them? or are computed using values of other options? or vales entered with a GUI by the user?

as I already said, XCDL, my build & configuration system, supports a much richer metadata to define these details, and I don't feel comfortable to give everything away simply because yotta is based on CMake and expects everyone to be happy with its limited features.

autopulated, 2015-11-24:
Happy to say that arbitrary code generation based on the config data is possible. This is something that's a current focus of our work to make mbed OS more portable :smiley: 

It's already used in mbed-hal: [these rules](https://github.com/ARMmbed/mbed-hal/blob/master/source/pins.cmake) use [this script](https://github.com/ARMmbed/mbed-hal/blob/master/source/pins.py) to generate the pin definitions. It's absolutely something we're looking for feedback on ways to improve though!

ilg-ul, 2015-11-24:
> ... feedback on ways to improve

from my experience with these things:

* Python is not the absolute solution (my first XCDL version was based on Python, and, although I used it a lot and enjoyed it, for serious projects I had to reconsider this); my next XCDL version will most probably be based on Go (the turning point was watching a presentation from Docker, having the same problems as I had).

* CMake is not the absolute solution. I'm currently taking a closer look at CMake, to better understand `yotta`, but my main complain is related to the syntax: `CMakeLists.txt` files were probably intended to be edited by humans, and are hard to parse/integrate in other build systems. I prefer my metadata to be stored in JSON files.

* if scripting snippets are needed in metadata files, my current recommendation is for JavaScript; it is currently supported quite well in most languages (C/C++, Java, Go for example); it is not perfect, but it is better than inventing a custom scripting/configuration language.

regards,

Liviu

autopulated, 2015-11-24:
You're probably right about python, and CMake: they are both compromises intended to make it possible to integrate smoothly with our internal tools (in the case of python), and with module authors who need a familiar way of describing arbitrarily complex builds (in the case of CMake).

CMake also had the advantage of an existing mechanism for describing multiple toolchains, tracking dependencies efficiently and generating IDE project files.

It certainly isn't ideal though!

ilg-ul, 2015-11-24:
> integrate smoothly with our internal tools (in the case of python)

yes, I understand, but try to convince hard core ruby users to install python and, even more, contribute some code to it.

so, indirectly, giving precedence to some internal reasons, you artificially limit acceptance for yotta. (as I said, _been there, done that_)

autopulated, 2015-11-24:
Thanks for the feedback about that. We do have a docker-containerised version of yotta, but I imagine that you can't assume people have the ability to run containers?

Out of interest: how would you feel about distributing it if yotta were written in node and packaged as a single executable using something like [`nexe`](https://github.com/jaredallard/nexe)? (though I suppose it would still need separate compiler/cmake/ninja executables to run)

ilg-ul, 2015-11-24:
> I imagine that you can't assume people have the ability to run containers?

that's correct

>  if yotta were written in node and packaged as a single executable

that would be an improvement over plain python, assuming the executable is statically built, or at least it is distributed with all libraries properly configured to run from the current folder, for osx, linux32, linux64, win32 and win64 (which is not an easy task, I do this when I repackage make, busybox, openocd, qemu).

but an Eclipse only solution would still be preferred, by far.