ARMmbed/yotta #591: Add custom CMake to be included by every module

bremoran, 2015-11-23:
Sometimes, a target needs to define custom CMake to be included by every module. For example, to support sensible code coverage, gcov should only be enabled for files that are under test, rather than all source files.

This implies that gcov needs to know about which files are being tested. The most sensible mechanism seems to be turning on coverage based on yotta config:

```JSON
"debug" : {
    "options" : {
        "coverage" : {
            "modules" : ["sockets", "sal"]
        }
    }
}
```

Then, custom CMake is needed to enable the coverage flags for source files within those modules. Since this custom CMake is specific to the target, it should live within the target. Adding custom CMake to each module would permit this.

cc @bogdanm @autopulated

bogdanm, 2015-11-23:
I understand the idea, but I'm not sure I understand the mechanism behind it. How would the config in the target "propagate" to the config in each module?

autopulated, 2015-11-23:
The mechanism I imagine is this:

 1. Target defines .cmake file(s) which will be included in the cmakelists for every module.

 2. These .cmake files can look at `${CMAKE_PROJECT_NAME}` and set target properties based on that, so for example:

```CMake
get_target_propertes(_compile_opts "${CMAKE_PROJECT_NAME}" COMPILE_OPTIONS )
list(APPEND _compile_opts "your options here")
set_target_propertes("${CMAKE_PROJECT_NAME}" PROPERTIES COMPILE_OPTIONS  "${_compile_opts}")
```

As the config is available at cmake-time, the .cmake files could add/set different properties based on the currently set config. Leaving aside dealing with the array in @bremoran's example, you could do:

```CMake
string(TOUPPER ${CMAKE_PROJECT_NAME} PROJECT_NAME_UPPERCASE)
if("YOTTA_CFG_DEBUG_OPTIONS_COVERAGE_INCLUDE_${PROJECT_NAME_UPPERCASE}")
   # should execute only if `debug.options.coverage.include.modulename` is truthy
endif()
```

bremoran, 2015-11-23:
There are two pieces to this approach.

1. Yotta needs to accept custom config on the command-line. Whether that's a file or a config string is up for debate. This allows me to add config like the above when building any module.
1. Yotta needs a way to apply compiler flags to some source files, but not others, without placing custom CMake in every module.

If I wrote a custom CMake file that does the following, it would be easy:

```CMake
if("${YOTTA_CFG_DEBUG_OPTIONS_COVERAGE_MODULES_${PROJECT_NAME}}")
    set(CMAKE_C_FLAGS_DEBUG_INIT        "${CMAKE_C_FLAGS_DEBUG_INIT} -fprofile-arcs -ftest-coverage")
    set(CMAKE_CXX_FLAGS_DEBUG_INIT        "${CMAKE_CXX_FLAGS_DEBUG_INIT} -fprofile-arcs -ftest-coverage")
    set(CMAKE_ASM_FLAGS_DEBUG_INIT      "${CMAKE_ASM_FLAGS_DEBUG_INIT} -fprofile-arcs -ftest-coverage")
    set(CMAKE_C_LINK_EXECUTABLE         "${CMAKE_C_LINK_EXECUTABLE} -fprofile-arcs -lgcov")
    set(CMAKE_CXX_LINK_EXECUTABLE         "${CMAKE_CXX_LINK_EXECUTABLE} -fprofile-arcs -lgcov")
endif()
```

This file then needs to exist in every project directory. That is where this issue comes in. This file lives in the target, but yotta processes it as though it appears in every target.

PR #633 (2015-12-18): Add the facility for targets to apply custom CMake to all modules.
 * Targets can define cmakeIncludes which will be included in every module
 * Make YOTTA_MODULE_NAME available to both CMake and the preprocessor, so that
   custom CMake and modules can use it.
 * Update documentation correspondingly

Fixes #591 and #626 

@bremoran @bogdanm: could you check that this suits your needs?
closed 2015-12-18 by autopulated