ARMmbed/yotta #592: Suggestion to use explicit json schema reference

ilg-ul, 2015-11-23:
I saw that you define a json schema, and validate files, against it, which is very good.

However, the json files you generate do not include any reference to the schema, or any versioning information, which, on long term, might be a problem.

In one of my projects I used an explicit `$schema` not only in the schema files, but also in the usual json files:

```
"$schema": "http://xcdl.github.io/schemas/xpack-1-1.json"
```

Even more, the URL is a real one, so even third party validators can download and use the schema. This is more or less in line with XML schema usage, where regular XML files usually refer to the schema.

Perhaps a similar approach would work for yotta too.

autopulated, 2015-11-24:
Thanks, this is certainly something that we should allow in the files so that editors which use it can provide schema-aware editing. The schemas are actually served by the registry at:

https://registry.yottabuild.org/schema/module
https://registry.yottabuild.org/schema/target

for convenience.

ilg-ul, 2015-11-24:
ok, that's fine. than only some versioning mechanism is needed, and adding a $schema reference in the json files.

btw, I agree that the schemas are to be consumed by machines, so formatting is not important, but... as a courtesy to those who read (possibly edit) them, the schemas would be even more appreciated if properly indented. I don't know how you guys work, but my json/c/c++ Eclipse editor has a reformat feature and I hit it after any edit.

autopulated, 2015-12-01:
+1 for versioning those endpoints, you're right.

I have a handle browser extension for auto-formatting json responses, so I didn't notice – but yes, those schemas are compressed before being sent, which kinda makes them a pain to read. This is because they're served the same way that other resources on the server are. I'll check if that's easy to change on a per-endpoint basis.

hugovincent, 2016-02-29:
+1 for explicit, versioned schema reference.