ARMmbed/yotta #597: Allow arrays in yotta config as unordered, unique sets

niklas-arm, 2015-11-24:
I have the following scenario: A controller-chip target inherits from a controller-series target (like STM32F429zi inherits from STM32F4).
The controller-series HAL defines all possible GPIOs on this series (something like PA_1 to PK_15) and the controller-chip target now wants to select only those that are actually available for itself.

This can be done already using:
```json
"pins": {
  "PA_1": true, 
  "PA_2": true, 
  "...": true,
  "PC_15": true
}
```

However, this is less verbose:
```json
"pins": [ "PA_1", "PA_2", "...", "PC_15"]
```

I understand that the semantics of inheriting and merging arrays is not well defined. Do we treat the array as ordered? Does it allow duplicates? Can values be removed from an array?

So I propose to treat arrays as unordered, unique sets of values set to true.
What are you thoughts on that?

autopulated, 2015-11-24:
So config lower down the inheritance hierarchy can only add things, not remove them?

niklas-arm, 2015-11-24:
Hm, yes?

If we were to treat the list as a pure alias to the true-dictionary, then you could remove individual items like this:
```json
"pins": {
  "PA_2": null
}
```

closed 2015-11-26 by niklas-arm