ARMmbed/yotta #601: yotta finishes build, but has non-zero exit code

markus-becker-tridonic-com, 2015-11-24:
When yotta finishes building successfully, but has non-critical version incompatibility errors in the install phase, it exits with a non-zero error code. This let's CI tools assume the build failed completely. Is this intended behaviour?

autopulated, 2015-11-25:
Yes, that's intended: yotta continues to try to build in the face of version errors because often they are non-fatal, but they are still considered errors. Hope that explains things!

markus-becker-tridonic-com, 2015-11-25:
Would be nice to have a --version-errors-not-fatal option.

mlnx, 2016-01-13:
@autopulated Is this something that could be considered for one of the short-term releases?

autopulated, 2016-01-13:
Version errors mean there was a failure, even if the build still succeeded, so I don't see this as high priority: is there a reason it should be?

markus-becker-tridonic-com, 2016-03-14:
There were quite a few version mismatches lately (mbed-drivers et.al.), which were not fatal, but made the CI system bail out on 'yotta install', so that a build was not even started.

markus-becker-tridonic-com, 2016-07-05:
Closing my yotta issues.

closed 2016-07-05 by markus-becker-tridonic-com