ARMmbed/yotta #603: repository url is confusing

niklas-arm, 2015-11-26:
In the module init yotta asks for the Repository URL, "where people can submit bugfixes".
That's what it says, but it wants the URI `git@github.com:user/repo.git` and that is definitely *not* the URL "where people can submit bugfixes".
```
Repository url (where people can submit bugfixes): https://github.com/user/repo
"https://github.com/user/repo" isn't a valid "repoObject" value.
```

Proposition:
```
Repository url (version control uri):
Homepage (where people can submit bugfixes):
```

jrobeson, 2015-11-26:
composer has an object/dict like this:

    email: Email address for support.
    issues: URL to the issue tracker.
    forum: URL to the forum.
    wiki: URL to the wiki.
    irc: IRC channel for support, as irc://server/channel.
    source: URL to browse or download the sources.
    docs: URL to the documentation.

An example:
```json
{
    "support": {
        "email": "support@example.org",
        "irc": "irc://irc.freenode.org/composer"
    }
}
```

autopulated, 2015-12-01:
"homepage" is intended to be for the "real" web-page homepage if there is one, which would normally be user docs etc., not necessarily the source repo. The plain github URL should be accepted in the repository field though: that can be improved.

There is also a separate bugs field you can fill in. The vast majority of modules will just have a single home (e.g. github/bitbucket) so I don't really think having lots of optional fields for different contact methods is useful. It's easy enough to put them in your readme if you want to, where they are easier for humans to find.

jrobeson, 2015-12-01:
I've found the docs field to be pretty useful, since lots of folks use readthedocs and similiar sites.