ARMmbed/yotta #607: Single-binary distribution

autopulated, 2015-12-01:
yotta has many and complex python dependencies, which can be a pain to install, especially on windows and Mac.

While we have a [windows installer](https://github.com/armmbed/yotta_windows_installer), [mac installer](), and in-development docker image (`docker pull yocker`) which can be used to install yotta more easily in some cases, IDE integration for yotta would be simpler if yotta was bundled as a single executable without python dependencies.

There are various ways of bundling python programs into a single executable: [py2exe](http://www.py2exe.org), [cx_freeze](http://cx-freeze.sourceforge.net), and [PyInstaller](http://www.pyinstaller.org). 

Some work needs to be done to test and pipe clean these processes, figure out which works best, and work-around or fix any issues in yotta or dependencies (for example resource loading from within a packaged executable will work differently).

thegecko, 2015-12-01:
Just an amendment, the docker image for yotta is ```mbed/yotta```, so to grab it use:

```docker pull mbed/yotta```

autopulated, 2015-12-01:
:+1: thanks

posborne, 2015-12-01:
FWIW, I have had great success using pyinstaller on a few projects lately.  As stated, you will need to make changes to support freezing.  When I introduced changes to modernize parts of pyocd, I did validate that they were able to build executables using these instructions: https://github.com/mbedmicro/pyOCD/blob/master/HOW_TO_BUILD.md (though some time has past since then).