ARMmbed/yotta #610: Using a private repository for targets

posborne, 2015-12-01:
I'm currently evaluating Yotta for use in building some of our embedded systems.  A majority of these systems are custom hardware (most likely based around a standard SoC).  Currently, there does not seem to be a way to use a Yotta target that is not published (without `target-link`).  For example, for a given project, I might want to instruct other developers to do the following:

    $ yotta target git+ssh://internal.git.server/project/target-custom-board.git

But this currently gives the following error:

     yotta.lib.access_common.AccessException: Target name "git+ssh://internal.git.server/project/target-custom-board.git" is not valid (must contain only lowercase letters, hyphen, plus, and numbers)

Putting this as the target in `.yotta.json` does not work either.  This seems like functionality that will be critical for use in real systems moving forward as customer do not and should not have any interesting in publicly releasing targets for their custom boards.

markus-becker-tridonic-com, 2015-12-02:
We are using the following .yotta.json 
```
{
  "build": {
    "target": "st-nucleo-f091rc-gcc,tridonic-com/st-nucleo-f091rc-gcc"
  }
}
```
to point to our target on GitHub. TO point to non-GitHub repos use the following syntax:

```
git@internal.git.server/project/target-custom-board.git
```

autopulated, 2015-12-02:
:thumbsup: that will work.

autopulated, 2015-12-02:
in general the same mechanism is used for downloading targets and modules, so the things here will work: http://yottadocs.mbed.com/tutorial/privaterepos.html except instead of specifying the source URL in "dependencies" of a module.json file, it needs to be specified where the target is set (either in the current directory's .yotta.json file, or in your global ~/.yotta/config.json file if there isn't a local one). You can also use this syntax in the inherits: property in target.json files.

I will leave this issue open to track the need for better documentation & a simpler way of doing this.

posborne, 2015-12-02:
Sounds good.  One issue I am running into now is that some URLs in our system use the tilde (`~`) character in paths which seem to be disallowed.

The examples in the documentation start with `git+ssh://` which does not seem to be allowed in this context.

autopulated, 2015-12-02:
OK, both of those sound like bugs. What are the error messages when you try to use ~ / git+ssh?

autopulated, 2015-12-02:
(if I set, for example:

`yotta target custom-board,git+ssh://internal.git.server/~project/target-custom-board.git`

or equivalently:

```json
{
  "build": {
    "target": "custom-board,git+ssh://internal.git.server/~project/target-custom-board.git"
  }
}
```

yotta correctly tries to access the git repo at git+ssh://internal.git.server/~project/target-custom-board.git 

Does it fail later than that?

posborne, 2015-12-02:
I think I was missing the target name as the first part of the comma-separated tuple before.  That was probably what was going wrong.

autopulated, 2015-12-04:
ok, that sounds like it would have caused the error you saw :+1:

posborne, 2015-12-04:
Indeed.  I was able to get this working as described last night when testing out the frdm-k22f-gcc target I added (published in the registry now as well): https://github.com/posborne/target-frdm-k22f-gcc