ARMmbed/yotta #611: define arbitrary scripts in module.json

jacobrosenthal, 2015-12-01:
As you probably know, npm has the concept of scripts defined in their package.json, This extensibility allows the community to create and test new functionality without having to necessarily upstream it. 
https://docs.npmjs.com/misc/scripts

For instance, Im not sure whats planned for uploading code, all the current docs say to drag and drop. 

Obviously a `cp build/target/source/blah.bin /Volumes/Mbed` would work nicely but it would be better if that was behind an `yt upload' which was an alias for `yt run upload`

This would seem related
https://github.com/ARMmbed/yotta/issues/234

Just curious what the temperature is on this. We the public dont get to see any roadmap talk so hard to know whats welcome.

jrobeson, 2015-12-02:
i asked for yt upload, but they didn't seem too keen on that.  One could use [mbed-ls](https://github.com/ARMmbed/mbed-ls) to achieve a (mostly) platform agnostic way to find the mounted mbed device

autopulated, 2015-12-02:
:+1: to more scripts, especially if it allows `yotta upload` or `yotta run` (possibly more appropriate name for non-cross-compiled targets)

jacobrosenthal, 2016-04-29:
Hey my PR #738 was closed as duplicate functionality, yet the PR that closed it specifically mentions my intent was not duplicate. Why would user defined scripts not be included especially when I've already written the code?

jacobrosenthal, 2016-04-29:
Seems like @autopulated thinks my only use case was yt run for module execution? I would like to ensure you thats not the case. For instance I have scripts that release code to an amazon s3 bucket and I in no way want that done every build on one of the preexisting hooks.