ARMmbed/yotta #612: "yotta whoami" should live-test OAuth authorizations

Patater, 2015-12-02:
Currently, "yotta whoami" does not live-test GitHub OAuth authorizations. It should live-test both GitHub OAuth and mbed OAuth authorization. Without this, there is no way to test cached OAuth credentials for validity before the first operation that requires OAuth credentials. The ability to test OAuth credentials for validity is important when one wants to determine if they should prompt for new credentials (in the event the credentials end up being invalid) when pre-populating the credentials cache.

To implement this feature, it may make sense to make a GET request to https://api.github.com/user with the cached OAuth token. The request, if authenticated, will return some information about the user. Otherwise, the request will contain `"message": "Requires authentication"`.

autopulated, 2015-12-02:
Note that even if the user has as valid login, they may not have a permission to access private repos that are being requested, so you should probably still handle the failure case (`yotta --noninteractive <subcommand>` can be useful for this, as it will prevent yotta hanging waiting for login).

Patater, 2015-12-02:
Yes, I agree. I currently use the "--noninteractive" feature.

The intent of the credential caching up front is to, in the case where the user sets up the cache with permission to access any necessary repositories or registries, ensure no prompts will be necessary in the successful workflow.

autopulated, 2015-12-02:
:thumbsup: got it