ARMmbed/yotta #625: yotta tries to create directory without name

markus-becker-tridonic-com, 2015-12-11:
In some cases (when we run yotta from a CI tool) yotta tries to create an directory without a name (empty string):
```
info: download mbed-client@tridonic from GitHub tridonic-com/mbed-client
info: get versions for mbed-client-c
Fatal Exception, yotta=0.11.0
Traceback (most recent call last):
  File "/usr/local/bin/yotta", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 205, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/install.py", line 62, in execCommand
    return installDeps(args, c)
  File "/usr/local/lib/python2.7/dist-packages/yotta/install.py", line 108, in installDeps
    test = {'own':'toplevel', 'all':True, 'none':False}[args.install_test_deps]
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 565, in satisfyDependenciesRecursive
    test = test
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 370, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 370, in __getDependenciesRecursiveWithProvider
    _processed = _processed
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 349, in __getDependenciesRecursiveWithProvider
    test = test
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 249, in __getDependenciesWithProvider
    satisfyDep, specs
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 231, in satisfyDep
    self.getName()
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 551, in provider
    r = access.satisfyVersionByInstalling(dspec.name, dspec.version_req, self.modulesPath())
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 294, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required, _registryNamespaceForType(type))
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 108, in latestSuitableVersion
    vers = remote_component.availableVersions()
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 485, in availableVersions
    return _listVersions(self.namespace, self.name)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 102, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 232, in _listVersions
    request_headers = _headersForRegistry(registry)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 410, in _headersForRegistry
    auth_token = generate_jwt_token(_getPrivateKeyObject(registry), registry)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 375, in _getPrivateKeyObject
    pubkey_pem, privatekey_pem = _generateAndSaveKeys(registry)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 348, in _generateAndSaveKeys
    settings.setProperty('keys', 'private', privatekey_pem.decode('ascii'))
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/settings.py", line 185, in setProperty
    set(section+'.'+name, value, save_locally)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/settings.py", line 182, in set
    parser.write(filename)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/settings.py", line 130, in write
    fsutils.mkDirP(dirname)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/fsutils.py", line 14, in mkDirP
    os.makedirs(path)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 2] No such file or directory: ''
```

autopulated, 2015-12-14:
Thanks for reporting this. I was able to reproduce by setting YOTTA_USER_SETTINGS_DIR to an empty string, could you confirm if that's the case for you?

closed 2015-12-14 by autopulated

markus-becker-tridonic-com, 2015-12-16:
Where is YOTTA_USER_SETTINGS_DIR set? Is this part of yotta config? How can I check?

autopulated, 2015-12-16:
It isn't normally set, but it can be set in order to prevent yotta from writing settings to a per-user directory (which is useful when running yotta in a CI system). I thought you might have set it after I mentioned it in response to [this issue](https://github.com/ARMmbed/yotta/issues/560), but if you don't have it set in your environment something else is going wrong to cause this problem!

markus-becker-tridonic-com, 2015-12-16:
It was indeed set to an empty value in the CI system. No idea, at which stage I put it there....

Thanks!

PR #627 (2015-12-14): Be more robust and handle failures when writing settings.
 * Log error instead of bailing due to OS errors (failing to write settings
   isn't really fatal)
 * Treat '' as '.', so if YOTTA_USER_SETTINGS_DIR is set to an empty string
   things don't break
 * Log the determined directory in a debug message to aid future debugging.

Fixes #625