ARMmbed/yotta #628: Registry search misses some results

jaustin, 2015-12-15:
When searching for modules, some of the results are missing or incorrect.
There's a module called bbc-microbit-classic-gcc however a search for this:
http://yotta.mbed.com/#search/bbc-microbit-classic-gcc
doesn't show the module. It does appear to erroneously show bbc-microbit-classic-armcc
[image]

Instad, searching for a substring from the module name:
http://yotta.mbed.com/#search/bbc-microbit-classic
..turns up the correct result

Doing the same game with bbc-microbit-classic-armcc *does* show all the results, so I wonder if this is the result of some overlapping metadata from copy/paste in the modules?

@jamessheppard @finneyj

autopulated, 2015-12-15:
There's a PR about to land on the registry will fix this.

The cause of this strange behaviour is that your search query is split into separate words based on the dashes, but the names of the modules it's being matched against aren't (currently) split in the same way.

autopulated, 2015-12-18:
Should be fixed now :+1:

closed 2015-12-18 by autopulated