ARMmbed/yotta #631: Dependency between changelog updates and versions

bremoran, 2015-12-16:
The workflow for changelogs isn't clear. For example, the most obvious workflow is:

1. update code
2. github Pull Request
3. yotta version [major|minor|patch]
4. update changelog
1. git commit -m "update changelog"
1. git push --tags origin master
5. yotta publish

The problem here is that the github tagged version won't have the same contents as the yotta registry.

The other possible workflow is:

1. update code
2. github Pull Request
4. update changelog
1. git commit -m "update changelog"
3. yotta version [major|minor|patch]
1. git push --tags origin master
5. yotta publish

In this case, there developer needs to guess what the version in the changelog needs to be.

In the ideal case, the version in the updated changelog will be contained within the git tagged version and the published version, but without breaking the semantics of ```yotta version```

autopulated, 2015-12-17:
I don't believe this is something yotta should be involved with. If you update the changelog immediately before doing `yotta version patch/whatever` then you can be sure what version you need to put in the changelog: there's no need to guess.

You could also use the notes on the github releases page to keep a changelog.

closed 2015-12-17 by autopulated

bremoran, 2015-12-17:
If that is the case, then yotta version patch/whatever is an unecessary API.

autopulated, 2015-12-17:
It's a a convenience for simple cases. It certainly won't cover all use-cases but I don't think that means it shouldn't exist!

I still use `yotta version <whatever>` after updating a changelog in order to edit the version and create a tag. Note that you can use `yotta version 1.2.3` to set an explicit version, if you're not sure that the result of `yotta version major/minor/patch` would correspond to your changelog.