ARMmbed/yotta #632: Yotta version should only output the version number not 'info: 1.3.1'

markus-becker-tridonic-com, 2015-12-16:
When trying to include the version number of the main module in CI, it would be beneficial to not get the superfluous 'info:'.

Is:
```
    $ yotta version
    info: 1.2.1
```
Expected:
```
    $ yotta version
    1.2.1
```

markus-becker-tridonic-com, 2015-12-16:
Note, this is not about ```yotta --version```.

autopulated, 2015-12-18:
This is an area I think we could make some other usability improvements too: sometimes people mistake `yotta version` and `yotta --version`. I notice that npm actually lists the versions of all its components (including the module in the current directory, if any) in the output of npm version.

How would you feel about formatted output including similar information, for example:

```sh
> yotta version
{
    "your-module-name": "1.2.3",
    "yotta": "0.12.0",
    "python": "2.7.11",
    "requests": "2.8.1",
    "PyGithub": "1.26.0",
    "cryptography": "1.0.2"
}

markus-becker-tridonic-com, 2015-12-21:
Definitely there is a usability issue with ```yotta version``` and ```yotta --version```.

I'd prefer a plain output, which one can immediately re-use, e.g in CI. How about ```yotta --plain version``` outputting ```1.2.1```?

markus-becker-tridonic-com, 2016-07-05:
Closing my yotta issues.

closed 2016-07-05 by markus-becker-tridonic-com