ARMmbed/yotta #637: Variable referenced before assignment error after updating to 0.12

LiyouZhou, 2015-12-18:
```
liyzho01@e105699-lin:/work/iot/nrf51-dfu-bootloader-ble$ yt build
info: get versions for mbed-hal-nrf51dk
info: download mbed-hal-nrf51dk@2.0.0 from the public module registry
info: generate for target: nrf51dk-gcc 1.0.0 at /work/iot/nrf51-dfu-bootloader-ble/yotta_targets/nrf51dk-gcc
Fatal Exception, yotta=0.12.0
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 205, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/build.py", line 46, in execCommand
    status = installAndBuild(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/build.py", line 120, in installAndBuild
    for error in generator.generateRecursive(c, all_components, builddir, application=app):
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/cmakegen.py", line 131, in generateRecursive
    c, all_components, os.path.join(modbuilddir, name), modbuilddir, processed_components, application=application
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/cmakegen.py", line 131, in generateRecursive
    c, all_components, os.path.join(modbuilddir, name), modbuilddir, processed_components, application=application
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/cmakegen.py", line 131, in generateRecursive
    c, all_components, os.path.join(modbuilddir, name), modbuilddir, processed_components, application=application
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/cmakegen.py", line 131, in generateRecursive
    c, all_components, os.path.join(modbuilddir, name), modbuilddir, processed_components, application=application
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/cmakegen.py", line 131, in generateRecursive
    c, all_components, os.path.join(modbuilddir, name), modbuilddir, processed_components, application=application
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/cmakegen.py", line 122, in generateRecursive
    self.generate(builddir, modbuilddir, component, new_dependencies, dependencies, recursive_deps, application, toplevel)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/cmakegen.py", line 501, in generate
    "module_is_empty": module_is_empty,
UnboundLocalError: local variable 'module_is_empty' referenced before assignment
```

jrobeson, 2015-12-21:
seems to have been introduced here: https://github.com/ARMmbed/yotta/commit/27568784a6b1928cbf138a0367454fe424c78843

autopulated, 2015-12-22:
:thumbsup:

autopulated, 2015-12-22:
(Note: before closing this, make sure that building with custom CMakeLists.txt is covered in tests, seems not to be!)

PR #639 (2015-12-22): Fix use of custom CMake
Ensure that use of custom CMake is covered by tests.

Resolves #637
commit 093e5773 (2015-12-22): fix #637
closed 2015-12-22 by autopulated