ARMmbed/yotta #640: yotta init command does not create sub-folder for placing the header files for applicaiton

maxpeng, 2015-12-27:
From the online documentation - [Your first application](https://docs.mbed.com/docs/getting-started-mbed-os/en/latest/FirstProjectmbedOS/), it describes that after executing *yt init* for the *blinky* application, *yotta* should create *blinky* sub-folder for placing the header files for our application. I just test it with *yotta v0.12.2*, it does not create *blinky* sub-folder. Is this the expected behavior?

autopulated, 2015-12-28:
I think that when you're creating an application (as opposed to a re-usable module) it makes sense to not create the public headers directory, as presumably your application doesn't expose a public API – though maybe that isn't true in the case that you will structure your application as something that you later split modules out of?

maxpeng, 2015-12-30:
I asked this because I wanted a module to include a header file provided by the application. Presently, the root directory of the application will be added to the searched directory of header files, however, it is better to let the application specify the include directory.
For example,
```
blinky
|-yotta_modules
|    \-module_a       ---> module_a needs to include module_a.h provided by the application.
|-source
     \-module_a_cfg
           \- module_a_cfg.h  ---> header file to customize the behavior of module_a.
```
Is it possible to do this?

autopulated, 2016-01-01:
Ah OK – thanks for the explanation!