ARMmbed/yotta #641: Yotta cmake + layout incompatible with existing libraries

ryankurte, 2015-12-28:
Hi there,

Looking at using some existing libraries with Yotta, however, unless I am missing something it is pretty difficult to make a library yotta compatible without significant alteration to the libraries.
It seems like it should be possible to add support without major refactoring, but I have run into a couple issues.

The first is that if you have any cmake files, they get automatically included. Any project using cmake will have a top level CMakeLists.txt file that may or may not (probably not) be yotta compatible, and there does not seem to be any way of ignoring this. According to the [docs](http://yottadocs.mbed.com/reference/buildsystem.html#custom-cmake) you can use a .yotta_ignore to ignore */CMakeLists.txt files, but not the top level file (or *.cmake files). So you have to remove it, which is a bit of a pain. This could be resolved by ignoring the CMakeLists.txt file and *.cmake files if specified in the .yotta_ignore file.

The next problem is that nothing matches the source/ module_name/ format, and reformatting the library to fit isn't sensible from a maintenance standpoint. The npm approach of specifying the source (and include, because c) folder(s) as part of the module would seem to resolve this.

Thanks for building awesome tools, very excited about yotta and mbed-os going forward.

Ryan

autopulated, 2015-12-28:
You should definitely be able to .yotta_ignore the top-level `CMakeLists.txt` file, so sounds like that's a bug!

I think it would be reasonable to specify the name of the source directory in `module.json` (applications already specify the directory that the binary should be built from, after all...), that it's currently forced to be `source` is something of a relic from when yotta was an internal-only tool trying to enforce some conventions on new code ;)

ryankurte, 2015-12-29:
Updated yotta to check (which I haven't done in a while, oops :-/) and nothing I do seems to ignore the cmake files. The other catch here is that they /always/ redefine project(), which doesn't work so well.
Put together a simple example [here](https://github.com/ryankurte/yotta-test}, in case I am doing something incorrect. Having a look at the yotta source, but it will probably take me a while to get my head around it.

I see you can take care of the headers with the extraIncludes field.

Also a note, I don't know whether it matters (but it took me an embarrasing while to stumble across, so maybe it will help someone else). If you are building a module for "libsodium", the module name in module.json must be "sodium", not "libsodium", otherwise the yotta_postprocess_target command in cmake will fail to infill the full library path and you get `ld: library not found for...`. Once that is right it's smooth sailing.

Thanks for the help!

autopulated, 2016-01-22:
(removing from milestone, and removing 'bug' tag as the remaining unresolved issue of being more flexible about the include directory is an enhancement, not really a bug)