ARMmbed/yotta #643: yotta corrupts sys.path, breaking its own assertions

ncoghlan, 2016-01-01:
As a side effect of importing yotta.lib, yotta inserts the lib directory into sys.path: https://github.com/ARMmbed/yotta/blob/master/yotta/lib/__init__.py#L6

This means that all subpackages of `yotta.lib` may now be imported under two different names, one with and one without the `yotta.lib` prefix.

I'm working on a Docker image for a [BBC:microbit build environment](https://github.com/ncoghlan/microbit/tree/master/devenv), and this is break (at least) the type assertion in `yt target`:

    $ yt target
    Fatal Exception, yotta=0.12.2
    Traceback (most recent call last):
      File "/home/microbitdev/.local/bin/yt", line 4, in <module>
        yotta.main()
      File "/home/microbitdev/.local/lib/python3.4/site-packages/yotta/main.py", line 205, in main
        status = args.command(args, following_args)
      File "/home/microbitdev/.local/lib/python3.4/site-packages/yotta/target.py", line 100, in execCommand
        return displayCurrentTarget(args)
      File "/home/microbitdev/.local/lib/python3.4/site-packages/yotta/target.py", line 66, in displayCurrentTarget
        args.target, c.targetsPath(), application_dir=app_path, install_missing=False
      File "/home/microbitdev/.local/lib/python3.4/site-packages/yotta/lib/target.py", line 155, in getDerivedTarget
        assert(isinstance(t, Target))
    AssertionError

Debugging that showed me that the result of `type(t)` was `<class 'target.Target'>`, while `Target` was `<class 'yotta.lib.target.Target'>` - the class object used to create `t` had come from a different instance of the module from the one used to create `Target`.

This kind of problem is going to be endemic as long as yotta is adding a package subdirectory directly to sys.path: http://python-notes.curiousefficiency.org/en/latest/python_concepts/import_traps.html#the-double-import-trap

autopulated, 2016-01-01:
:+1: 

This is a real problem in the way things are packaged. I think fixing it might be possible just by using absolute imports (yotta.lib.xxx) everywhere.

I don't understand why this assertion occurs in your environment and none other though.

autopulated, 2016-01-20:
This should have been fixed by #670, which will be included in a published release within the next week. If you still experience any problems please re-open!

closed 2016-01-20 by autopulated

ncoghlan, 2016-01-21:
Thanks! I'll respin https://hub.docker.com/r/ncoghlan/microbit-dev/ and try the `yt target` command again once the new release is out.

(fortunately `yt up`, `yt build` and `yt logout` weren't affected, so this didn't prevent putting together a usable Docker-based build environment).

PR #670 (2016-01-20): always use full path to import subpackages, don't mess with sys.path
should fix #643
commit c710ffbf (2016-01-20): always use full path to import subpackages, don't mess with sys.path; should fix #643