ARMmbed/yotta #647: Cosmetic: `yt target nonesuch` crashes with Python 3

mpg, 2016-01-07:
With Python 2 I get a clean and informative error message as expected. However, on a machine where /usr/bin/python is Python 3, I get:

```
info: get versions for nonesuch
Fatal Exception, yotta=0.12.2
Traceback (most recent call last):
  File "/usr/bin/yt", line 4, in <module>
    yotta.main()
  File "/usr/lib/python3.5/site-packages/yotta/main.py", line 205, in main
    status = args.command(args, following_args)
  File "/usr/lib/python3.5/site-packages/yotta/target.py", line 131, in execCommand
    target, errors = c.satisfyTarget(t)
  File "/usr/lib/python3.5/site-packages/yotta/lib/component.py", line 581, in satisfyTarget
    additional_config = additional_config
  File "/usr/lib/python3.5/site-packages/yotta/lib/target.py", line 149, in getDerivedTarget
    (dspec, previous_name)
TypeError: __str__ returned non-string (type bytes)
```

autopulated, 2016-01-07:
:thumbsup: Thanks for reporting!

autopulated, 2016-01-20:
Fixed by #671

closed 2016-01-20 by autopulated