ARMmbed/yotta #649: support for specifying target in build command

pelikhan, 2016-01-08:
We would like to be able to run and specify the target in the same command:

    yotta build --target <sometarget>

jrobeson, 2016-01-08:
you hve it backwards.  yotta -t/--target <target> build works

pelikhan, 2016-01-08:
Say I have

    yotta target TARGET
    yotta build

How can I get it into a single command? Below does not work:

    yotta --target TARGET build

jrobeson, 2016-01-08:
works for me right now on the current yotta version. 

```sh
$ yotta -t nrf51dk-gcc build                             
info: generate for target: nrf51dk-gcc 1.0.0 at project/yotta_targets/nrf51dk-gcc
GCC version is: arm-none-eabi-g++ (Fedora 5.2.0-3.fc23) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

suppressing warnings from nrf51-sdk
suppressing warnings from ble-nrf51822
-- Configuring done
-- Generating done
-- Build files have been written to: project/build/nrf51dk-gcc
[2/2] Linking CXX executable source/project
Memory usage for 'project'
section              size
.data                 144
.bss                 2800
.heap               17536
.stack               2048

```

jrobeson, 2016-01-08:
and here's an example from the same directory with a different traget, notice the change in build directory

```sh
```sh
$ yotta -t blenano15-gcc build 
info: generate for target: blenano15-gcc 1.0.0 at project/yotta_targets/blenano15-gcc
GCC version is: arm-none-eabi-g++ (Fedora 5.2.0-3.fc23) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

suppressing warnings from nrf51-sdk
suppressing warnings from ble-nrf51822
-- Configuring done
-- Generating done
-- Build files have been written to: project/build/blenano15-gcc
[1/1] Linking CXX executable source/project
Memory usage for 'project'
section              size
.data                 144
.bss                 2800
.heap               17536
.stack               2048

```
```

autopulated, 2016-01-08:
Yup `yotta --target <targetname> build` should work – If you're seeing an error with that please post the output!

pelikhan, 2016-01-08:
Is it possible to specify this command the other way?

    yotta build --target <targetname>

Trying to get yotta to run automatically from Visual Studio code but it does not like options before the task name.

jrobeson, 2016-01-08:
sounds like a bug in visual studio code then.

pelikhan, 2016-01-08:
It is designed for tools like node, jake, grunt, ... that usually take the task name as first argument.

Why is there an order requirement?

jrobeson, 2016-01-08:
I can't answer that one, so I'm bowing out.  (I don't actually work on yotta)

autopulated, 2016-01-10:
I recently did some work to make it easier to add options after subcommand names, I agree that would be desirable.

(Because of the way python's argparse subcommand parsers work, it was originally simpler and required much less duplication to have the shared options at the top level)

I'm leaving this open to track as a bug the fact that --target should be usable after the subcommand name too.