ARMmbed/yotta #652: yotta linked module doesn't get included in the build process

jrobeson, 2016-01-12:
I created a local module called lbcommon and ran `yotta link`. 
I then went to the module I want to use the linked module in and ran `yt link lbcommon`.  

I see the symlink in `yotta_modules/`. I then added `#include lbcommon/util.h"` to my `source/main.cpp` 

I then ran `yotta build` and got this error:

`~/projects/test/source/main.cpp:8:29: fatal error: lbcommon/util.h: No such file or directory`

I looked into `build/nr51dk/ym` and do not see my linked module there.  Should I be seeing it there?

NOTE: I do override `YOTTA_PREFIX` to use `~/.local/share/yotta/` instead of the default in `/usr/local`.

That may or may not be related ot the problem

jrobeson, 2016-01-12:
i just tested with `YOTTA_PREFIX` unset , and still have the same problem

jrobeson, 2016-01-12:
ahh.. of course you still have to set the module as your dependency!

closed 2016-01-12 by jrobeson

autopulated, 2016-01-12:
:wink: