ARMmbed/yotta #661: Installation issues on Ubuntu 14.04

autopulated, 2016-01-15:
There are some conflicts between yotta's dependencies and the versions required by system tools on Ubuntu 14.04: either we should recommend using a virtualenv to install, or relax the specifications on dependencies if appropriate.

cc @janjongboom

ddavidebor, 2016-01-21:
absolutely!

Today this same error emerged to me on ubuntu 14.04:

https://bugs.launchpad.net/openstack-ansible/+bug/1535998

It's related to pip 8.0.0 conflicts.

autopulated, 2016-01-22:
I think the pip 8.0.0 issues are a separate problem, which I believe pip 8.0.1 has now fixed (`pip install -U pip>=8.0.1`). (This issue was reported before pip 8.0.0 was released)

autopulated, 2016-01-22:
I've tested the updated instructions in https://github.com/ARMmbed/yotta/pull/673, which will be merged shortly, and I can't reproduce any problems now. Please open a new issue with specific errors if problems are encountered again!

closed 2016-01-25 by autopulated