ARMmbed/yotta #666: yotta update --patch-only / --minor-only

autopulated, 2016-01-19:
(--patch-only/--minor-only names up for discussion)

It would be useful to be able to add an additional restriction on the versions of modules to update to. For example, you might only want to pull in bugfix (patch) updates, and not new features, even if the version specifications in modules allow minor updates.

janjongboom, 2016-01-19:
If the modules use semver, you could specify it as "module-name": "0.3.x" right? I dunno if yotta understands that syntax though.

autopulated, 2016-01-19:
In `module.json` you can (and should!) specify a range of compatible versions, (although the 1.2.x syntax isn't currently supported): http://yottadocs.mbed.com/reference/module.html#dependencies

This would be to apply an additional restriction: so that even if modules specify in their dependency lists that minor (new feature) updates are OK, you as an application developer have the choice to pull in only bugfix updates.