ARMmbed/yotta #669: error: server returned status 400: {"error":"jwt has expired"}

PrzemekWirkus, 2016-01-20:
Got this issue already for some time. I'm sure that I'm doing something wrong but still, I can't use yotta with my Linux box (```Linux mbedci 3.13.0-62-generic #102-Ubuntu SMP Tue Aug 11 14:29:36 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux```).

# Description
When I try to build ```mbed-drivers``` on my Linux machine I got ```error: server returned status 400: {"error":"jwt has expired"}``` error from yotta.

# Example
```
$ cd mbed-drivers
$ yt --version
0.13.0
$ yt target
frdm-k64f-gcc,* missing
$ cat module.json | grep version
  "version": "0.11.8",
```

```
yt up
info: get versions for frdm-k64f-gcc
info: download frdm-k64f-gcc@2.0.0 from the public module registry
error: server returned status 400: {"error":"jwt has expired"}
Fatal Exception, yotta=0.13.0
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 206, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/update.py", line 37, in execCommand
    target, errors = c.satisfyTarget(args.target, update_installed=True, additional_config=args.config)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 633, in satisfyTarget
    additional_config = additional_config
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/target.py", line 143, in getDerivedTarget
    type = 'target'
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 370, in satisfyVersion
    name, version_required, working_directory, type=type, inherit_shrinkwrap = inherit_shrinkwrap
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 306, in satisfyVersionByInstalling
    name, version_required, install_into, v, type=type, inherit_shrinkwrap = inherit_shrinkwrap
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 317, in _satisfyVersionByInstallingVersion
    version.unpackInto(working_directory)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 447, in unpackInto
    _getTarball(self.url, directory, self.sha256)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 102, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 185, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 168, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 149, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 289, in _getTarball
    response.raise_for_status()
  File "/usr/local/lib/python2.7/dist-packages/requests/models.py", line 840, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 400 Client Error: Bad Request for url: https://registry.yottabuild.org/targets/frdm-k64f-gcc/versions/2.0.0/tarball
```

autopulated, 2016-01-20:
The clock on your box is probably set incorrectly.

yotta uses signed JSON Web Tokens to authenticate requests against the registry, these have an expiration time set to reduce the possibility of replay attacks.

I notice that even though you sent this bug shortly after 14:00 GMT, your clock reports: "14:29:36 GMT" as the current time, so this seems to support that explanation.

PrzemekWirkus, 2016-01-20:
Spot on James! :)
```
$ sudo ntpdate -s ntp.ubuntu.com
```
My machine's clock was off by 4-5 minutes.

autopulated, 2016-01-20:
:thumbsup:

bridadan, 2016-01-20:
It's worth mentioning that @geky ran into this issue too. He figured it out, but since this seems to be recurring maybe it makes sense to add a warning? Probably something like:

```
error: server returned status 400: {"error":"jwt has expired"}
warning (hint?): Is your system clock accurate?
```

autopulated, 2016-01-20:
Good point, I will leave this open to track the need for a more friendly error message.

urbanthings, 2016-01-25:
I get the same error & machine clock seems just fine

alteredworlds, 2016-01-25:
I get the same even though machine clock fine

alteredworlds, 2016-01-25:
FYI: I'm going through 'blinky' after initial install of Docker on Mac OS X 10.11.3. I got the above error multiple times in a row until I created the 'blinky' (empty) header folder. After that, I was properly shown 'You need to log in to do this.' & have been able to continue.

autopulated, 2016-01-25:
Thanks for reporting that. Are you using the yotta docker image that we provide, or your own?

@thegecko could there be an issue with clock synchronisation when the docker image first boots up?

alteredworlds, 2016-01-25:
I followed the instructions at:
https://docs.mbed.com/docs/getting-started-mbed-os/en/latest/docker_install/ <https://docs.mbed.com/docs/getting-started-mbed-os/en/latest/docker_install/>

thegecko, 2016-01-26:
I've not come across this before, but that doesn't mean it isn't a problem :)
The interwebs tell me that you can mount the following folder when running to ensure the clock is synced:

```-v /etc/localtime:/etc/localtime:ro```

@alteredworlds, I'd be interested to know if adding this to the yotta.sh file resolves your issue.

alteredworlds, 2016-01-26:
With regards to further diagnosis of this issue, I'm afraid I can no longer replicate it. The error occurred only on first access to the yotta registry (when not yet logged in). I was able to get past the error, *apparently* via some side effect of manually creating the header folder, but correlation != causation... :)

Hokins-Liu, 2016-04-08:
I got the same issue but I use yotta in win7, and i checked the system time is synced with the internet time setting, is anyone know how to fix this issue in win7 system?

P.S. this issue cause the yotta can not login account, install library, device target, etc...

autopulated, 2016-04-12:
If you're getting this error it means your clock isn't synchronised correctly. Most likely this might be because it's set to the wrong time-zone. Note that it needs to be correct within a few minutes – choosing the option to automatically set the time is the best way to do that.

Hokins-Liu, 2016-04-12:
Hi @autopulated , I think the clock is synchronised currectly, and the time zone is setup the right setting but still no work, so, may it could be another reason to cause the issue appears in my system.

autopulated, 2016-04-12:
That's odd – please could you check that the output of running: `python -c "import datetime; print(datetime.datetime.utcnow())"` in the console matches the time displayed, for example, here: http://time.is/UTC

If python's notion of time isn't what your system clock is showing, that could also cause this problem

Hokins-Liu, 2016-04-13:
Hi @autopulated , it looks like delayed about 3~4 minutes, 
and then manual modify the system time could be done for yotta login and get other library depencies

Thanks.

autopulated, 2016-04-13:
Thanks for the update – just to clarify, is the time shown by python 3-4 minutes off the time shown by your system clock, or is the time shown by your system clock 3-4 minutes off the http://time.is/UTC time?

Hokins-Liu, 2016-04-13:
it's the system clock 3-4 minutes off the http://time.is/UTC time.

autopulated, 2016-04-13:
ah ok – that'd be enough to invalidate the JSON web tokens used for authentication.

I think the ideal solution to this problem would be for the yotta client to get the time used for authentication from the registry. The authentication scheme is designed to work in the clear (even though requests are always made over https), so getting the time securely needs some careful thought.

Hokins-Liu, 2016-04-13:
I think sometimes it could get some confused in old PC, some PC system time can not be synchronized with UTC time setting, and they just change the time by manual modification (like the case of me). 
But it may not be cool to do that for someone just using yotta in first time.
actually I installed yotta tool already 5 PCs, it is the first time I met the issue, just get a little shock about the error coming.
Give the friendly message is cool for user experience, I think it's really good for first time using.

Still thank you for your help, @autopulated !