ARMmbed/yotta #672: Include paths are not passed to the compiler properly when using CMake >=3.4

Patater, 2016-01-21:
When building a yotta project with CMake of version 3.4 (or 3.4.1), error messages about missing include files appear.

For example, when building minar, we get `fatal error: uvisor-lib/uvisor-lib.h: No such file or directory`.

One can see the `-I` for all dependent yotta modules is missing on the command line. For example, `-I /path/to/yotta_modules/uvisor-lib` is missing.

Downgrading CMake to 3.3.2 is a workaround.

The CMake 3.4 release notes say:

>The internal CMAKE_\<LANG\>_COMPILE_OBJECT rule variable now substitutes compiler include flags in a separate \<INCLUDES\> placeholder instead of the main \<FLAGS\> placeholder.

autopulated, 2016-01-22:
:thumbsup: Thanks for reporting this!

autopulated, 2016-01-25:
(the fix will be in the two targets referenced above, so this issue will not block the 0.13.1 release of yotta, which should be later today)

autopulated, 2016-01-29:
This issue was actually resolved in an earlier version, the report turned out to be due to a derived target still inheriting from old versions of mbed-gcc/armcc.

closed 2016-01-29 by autopulated