ARMmbed/yotta #675: Don't put links of modules and targets in '/usr/local'

bearsh, 2016-01-22:
Let's cite FHS:
"/usr is the second major section of the filesystem. /usr is shareable, read-only data. That means that /usr should be shareable between various FHS-compliant hosts and must not be written to. Any information that is host-specific or varies with time is stored elsewhere."
This of course also applies to any folder under '/usr'.
I assume '/var/lib' would be to right place to put these links in.
It would also be nice to have an option for the link and link_target command to use a place in the users home directory (like in ~/.config/yotta) without fiddling with the YOTTA_PREFIX env var. In my opinion, it would be even better to have the home directory the default and a '-g' option to link in globally...

autopulated, 2016-01-25:
Defaulting to the user's home directory might well be the best thing to do.

While the directory probably should change, for now the best thing to do would be set `YOTTA_PREFIX` in your shell's startup file, and then forget about it.

I don't think command-line options for the prefix make sense, because it isn't something that you could change between commands and have them function correctly (unlike say `--target`). That said, the ability to use one module as a dependency of another directly, without an intermediate installation directory, might be what you're really after here?

bearsh, 2016-01-25:
Well I didn't mean to change the prefix with a command-line option. It's more about where to store the link. If I call `link[_target]` to create a link, it should be created in the uses home directory. But if I call link with the `-g` option, the link should be created in `YOTTA_PREFIX/` if it exists, if not in `/var/lib/yotta`. If I call `link[_target]` to use a link, it should search through the users config and the global link place.

looking a the link.py and link_target.py, it should be fairly easy to change it to default to the users home (`userSettingsDirectory()` event exists in folders.py). and the `-g` option makes it still usable for everyone which already installed its links globally. changing the global config directory to a **sane** place could be done in a second step.

mjs-arm, 2016-02-24:
> I don't think command-line options for the prefix make sense, because it isn't something that you could change between commands and have them function correctly (unlike say --target). 

Why would this not work?  Is it not just a case of providing the extra option such that the link command knows where to find the component?

The current mechanism makes it awkward to create to parallel development areas where each uses the same component at different semver.  (Consider bug fixing scenarios on different major.minors)