ARMmbed/yotta #683: What is the dependency order resolution in yotta?

javier-moreno-tridonic-com, 2016-01-26:
It looks like it is "Breadth-first" going first through the dependencies and then the target dependencies of the executable module in the order they appear in the module.json.

But based on my experience when we go down a level, the order in the module.json does not matter. So, for example if my module.json looks like the following:
```
...
"dependencies": {
    "A": "1.0.0",
    "B": "1.0.0"
 },
"targetDependencies": {
    "myTarget": {
      "C": "1.0.0"
    }
...
```
I could observe cases were **yotta install** will result in:

> fetching A
> fetching B
> fetching C
> fetching A dependencies
> fetching C dependencies
> fetching B dependencies

Is there a way to predict in which order the modules will get fetched?

mlnx, 2016-01-26:
cc @mlnx

autopulated, 2016-01-26:
**Short answer:** it's deterministic, but unspecified. It may change in the future, so don't rely on it, but any differences in the order between the same version of yotta running on a different system are considered a bug.

**Long answer:** A mix of breadth first and depth-first: First the immediate dependencies are all installed in a breath-first fashion. The order is `dependencies` then `targetDependencies` then `testDependencies` (if appropriate). If a dependency occurs multiple times within the resolved dependencies of a single module, the first occurrence is processed and the others are ignored. The dependencies of each of the installed modules are then resolved **recursively** (depth-first) following the same rules, before proceeding to the next.

In your case I suspect A also has a dependency on C, which is what causes C's dependencies to be fetched before B's.

In the medium term, this may well change to a pure bredth-first order, not least because it would make it easier to parallelise the download of modules. In the long term, it is likely that we will solve for version conflicts with a SAT solver (see https://github.com/ARMmbed/yotta/issues/537). I think if we do that successfully, the installation order should be irrelevant, as exactly what versions are being installed will have been resolved prior to any modules actually being written to disk.

@mlnx you can watch an issue using the "Subscribe" button in the sidebar on the right, no need to comment ;)

javier-moreno-tridonic-com, 2016-04-25:
Thanks for the answer, closing the ticket.

closed 2016-04-25 by javier-moreno-tridonic-com