ARMmbed/yotta #688: pip install -U yotta fails with older version of project_generator

sg-, 2016-01-29:
Upgrading yotta 0.8.3 to 0.13.1 found the following:

```
Collecting hidapi (from pyOCD<1.0,>=0.3->valinor<1.0,>=0.0.0->yotta)
  Downloading hidapi-0.7.99.post12.tar.gz
    Complete output from command python setup.py egg_info:
    usage: pgen [-h] [-v] [-q] [--version]
                {list,init,export,build,clean,import} ...
    pgen: error: invalid choice: '/tmp/easy_install-2OlQmk/Cython-0.23.4/Cython/Parser/Grammar' (choose from 'list', 'init', 'export', 'build', 'clean', 'import')
    Traceback (most recent call last):
      File "<string>", line 20, in <module>
      File "/private/tmp/pip-build-VDlJve/hidapi/setup.py", line 90, in <module>
        **setup_args
      File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/dist.py", line 268, in __init__
        self.fetch_build_eggs(attrs['setup_requires'])
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/dist.py", line 313, in fetch_build_eggs
        replace_conflicting=True,
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pkg_resources/__init__.py", line 836, in resolve
        dist = best[req.key] = env.best_match(req, ws, installer)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1081, in best_match
        return self.obtain(req, installer)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1093, in obtain
        return installer(requirement)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/dist.py", line 380, in fetch_build_egg
        return cmd.easy_install(req)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 638, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 668, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 851, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1079, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1065, in run_setup
        run_setup(setup_script, args)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/sandbox.py", line 240, in run_setup
        raise
      File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/sandbox.py", line 193, in setup_context
        yield
      File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/sandbox.py", line 164, in save_modules
        saved_exc.resume()
      File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/setuptools/sandbox.py", line 138, in resume
        type, exc = map(pickle.loads, self._saved)
      File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 1382, in loads
        return Unpickler(file).load()
      File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 858, in load
        dispatch[key](self)
      File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 1133, in load_reduce
        value = func(*args)
    TypeError: __init__() takes at least 3 arguments (1 given)
```

and it seemed the problem was name collision
```
Collecting hidapi (from pyOCD<1.0,>=0.3->valinor<1.0,>=0.0.0->yotta)
  Downloading hidapi-0.7.99.post12.tar.gz
    Complete output from command python setup.py egg_info:
    usage: pgen [-h] [-v] [-q] [--version]
                {list,init,export,build,clean,import} ...
    pgen: error: invalid choice: '/tmp/easy_install-2OlQmk/Cython-0.23.4/Cython/Parser/Grammar' (choose from 'list', 'init', 'export', 'build', 'clean', 'import')
```
project_generator version 0.7.3 used pgen as the alias. Solution was to uninstall project_generator and then update yotta.
```
$ pip uninstall project-generator
$ sudo pip install -U yotta
```
Anyone know if you can force uninstall modules in setup.py with something similar to 'install_requires' or in another way?

autopulated, 2016-02-01:
Nope, sorry, don't think that's possible.

The fix for this is to re-install cython (replacing the pgen executable with its own version again). When project_generator was installed this executable was clobbered, there's no way to get it back and fix make installation work again without re-installing cython.

I don't think there's anything we can do about this in yotta, so closing here.

sg-, 2016-02-01:
It's pretty bad if we have dependencies that break things that then don't allow us to update our own tools. There has to be something mitigate this behavior. Its a horrible experience :-1:

autopulated, 2016-02-01:
@sg- this issue was fixed upstream in October (https://github.com/project-generator/project_generator/issues/271).

I don't believe it's possible to mitigate it, but if it is I'm sure https://github.com/project-generator/project_generator would appreciate a pull request to do that for all projects that depend on it, and that would be the right place to make such a fix.

closed 2016-02-01 by autopulated

sg-, 2016-02-01:
The problem is that the damage is done. Not sure there is anything to fix in project_generator. In general deploying production tools with such flaws (cannot uninstall dependencies before installing) is mind bending.

0xc0170, 2016-02-04:
> I don't believe it's possible to mitigate it, but if it is I'm sure https://github.com/project-generator/project_generator would appreciate a pull request to do that for all projects that depend on it

How can I help?