ARMmbed/yotta #689: Build errors due to dependency issues do not stop the build process

mlnx, 2016-02-01:
```
> yt build
.
.
error: mbed-client-c does not meet specification ^1.0.0 required by sal-stack-nanostack
error: mbed-drivers does not meet specification ~0.11.8 required by mbed-6lowpan-eventloop-adaptor
.
.
.

[385/385] Linking CXX executable source/mbed-client-example-6lowpan
```

A binary is generated even if there are dependency issues as shown above.

Use this repo to replicate problem:
https://github.com/ARMmbed/mbed-client-example-6lowpan (device_server branch)

yotta version: 0.13.0

autopulated, 2016-02-02:
This is by design. In this case yotta's exit-status is non-zero, so CI systems can still pick up that there was a failure. On the command line you could check the exit status with, for example:

`yotta build || echo "build failed (non-zero status)"`

closed 2016-02-02 by autopulated

mlnx, 2016-02-02:
I still think this doesn't make sense. If there are errors, you would expect the image not to be generated.
It's not clear which modules/versions are being used in the image.

autopulated, 2016-02-02:
If you like, you can pretend that the image is not generated if there are errors ;)

The modules used will always be those listed by `yotta ls`, this should also show you all the version errors, or if it does not, try `yotta ls -a`.

bridadan, 2016-02-02:
Would it make sense instead of labeling this as an `error` to label this as a `warning`? Since that's what it seems like this actually is. yotta has already fetched the modules, but it has realized there is an incompatibility according to the version numbers. No guarantee that it will fail, no guarantee that it will work, just warn that there's definitely a point of failure

autopulated, 2016-02-02:
It's an error because you really really should fix it, and I think people would be too inclined to ignore a warning – otherwise it would be a warning

mlnx, 2016-02-02:
cc @janjongboom @maclobdell 

correct! Therefore by default do not generate an image, as it's very confusing. We spent lots of time fiddling with dependencies and trying to fix the issues, until we realized that the image was there.
The user experience is not good here.

For the CI system, you could have a switch option to ignore dependency issues.

autopulated, 2016-02-02:
> We spent lots of time fiddling with dependencies and trying to fix the issues

Good! You should have done :wink: 

> until we realized that the image was there

You're suggesting to remove the image for everyone, because *you* decided that it's OK to ignore error messages if an image is created, but for some reason wish that you hadn't?

Let me be clear: the fact an image is produced is *not for the benefit of a CI system* (the *exit status* indicating whether there was an error is for the benefit of a CI system), the fact an image is produced is for the benefit of the user for whom we want to maximise the possible chance of a successful build that they can use to test whatever they're working on, even in the presence of other errors.

Perhaps this is a product of my own approach to software development, where I want something I can run and test as soon as possible, and then incrementally improve. (Instead of constructing some grand and complete edifice before first attempting to run it). But as far as software development is concerned, I believe this is a wise practise to encourage.