ARMmbed/yotta #693: yt link-target should either set the target or remind me to

jaustin, 2016-02-02:
Behaviour repeatedly observed, including me who should know better:
```
(local-target-dir )$ yt link-target #target is called my-target in module.json
(my-binary-module)$ yt link-target my-target
(my-binary-module)$ yt target
**Not the target I just linked**
```
Oh no, something's broken...
...
...

eventually
...
```
(my-binary-module)$ yt target my-target
```
And now it all works

Are there good reasons that yt link-target doesn't also set the target? I assume so... therefore:
Could yt link-target please also say 
```
Target linked. You must run:
yt target target-name
...to set the target locally
```

autopulated, 2016-02-03:
:thumbsup:

autopulated, 2016-03-24:
This was fixed by #709

closed 2016-03-24 by autopulated