ARMmbed/yotta #695: Fatal Exception, yotta=0.13.1

david-reimondez-tridonic-com, 2016-02-03:
I have just installed Yotta following the manual procedure described in the website in a windows 7 machine.

It doesn't matter what command I execute I always get this message error 

$ yt --version
Fatal Exception, yotta=0.13.1
Traceback (most recent call last):
  File "C:\Python27\Scripts\yt-script.py", line 9, in <module>
    load_entry_point('yotta==0.13.1', 'console_scripts', 'yt')()
  File "C:\Python27\lib\site-packages\yotta\main.py", line 76, in main
    options.target.addTo(parser)
  File "C:\Python27\lib\site-packages\yotta\options\target.py", line 24, in addTo
    default=detect.defaultTarget(),
  File "C:\Python27\lib\site-packages\yotta\lib\detect.py", line 14, in defaultTarget
    set_target = settings.getProperty('build', 'target')
  File "C:\Python27\lib\site-packages\yotta\lib\settings.py", line 174, in getProperty
    return get(section + '.' + name)
  File "C:\Python27\lib\site-packages\yotta\lib\settings.py", line 169, in get
    _ensureParser()
  File "C:\Python27\lib\site-packages\yotta\lib\settings.py", line 153, in _ensureParser
    parser.read(config_files)
  File "C:\Python27\lib\site-packages\yotta\lib\settings.py", line 66, in read
    self.configs[fn] = ordered_json.load(fn)
  File "C:\Python27\lib\site-packages\yotta\lib\ordered_json.py", line 20, in load
    return json.load(f, object_pairs_hook=OrderedDict)
  File "C:\Python27\lib\json\__init__.py", line 290, in load
    **kw)
  File "C:\Python27\lib\json\__init__.py", line 351, in loads
    return cls(encoding=encoding, **kw).decode(s)
  File "C:\Python27\lib\json\decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "C:\Python27\lib\json\decoder.py", line 382, in raw_decode
    obj, end = self.scan_once(s, idx)
ValueError: Expecting , delimiter: line 10 column 3 (char 2365)

autopulated, 2016-02-04:
It looks like your yotta settings file (probably `C:/Users/<your-user-name>/.yotta/config.json`) could have become corrupted somehow.

In any case this should not cause such a fatal error, but for now you could try removing your settings file.

yotta can read settings from several different places, so to find out exactly which file yotta is attempting to read when this error occurs, you could edit the section of `C:\Python27\lib\site-packages\yotta\lib\settings.py` around line 66 to read:

<pre>
        for fn in filenames:
            try:
                <b>logging.info("reading settings from %s", fn)</b>
                self.configs[fn] = ordered_json.load(fn)
            except IOError:
                self.configs[fn] = OrderedDict()
</pre>

david-reimondez-tridonic-com, 2016-02-04:
Thanks a lot!

solved!

autopulated, 2016-02-04:
No problem – I'm going to leave this issue open to track the need for better handling of this class of error (yotta should at least give you a nice message saying that the syntax in the config file was invalid, and not an ugly backtrace!)

PR #755 (2016-03-29): handle syntax errors in settings files more gracefully
Fixes #695
closed 2016-03-29 by autopulated