ARMmbed/yotta #698: Allow config prefix to be set based on config path

bridadan, 2016-02-05:
Currently for mbed-hal, we generate "short names" for compatibility. This is done through a python script in mbed-hal's source: https://github.com/ARMmbed/mbed-hal/blob/master/source/pins.py

Modifying yotta config prefixes is not a unique need for mbed-hal, so perhaps we could create a mechanism within yotta for this.

In the root of the target/module.json object, we could add a new key. I will call this `configPrefixes".

It could potentially look this:

```
{
  ...
  "config": {
    "hardware": {
      "pins": {
        "LED_RED":  "PTB22",
        ...
      }
    }
  },
  "configPrefixes": [
    {
      "path": ["hardware", "pins"],
      "prefix": ""
    },
    ...
  ]
}
```
I would imagine this would be a fairly simple PR, mostly looking for feedback and discussion on this.

autopulated, 2016-02-08:
I'm not convinced that it's a good idea to expose yotta config values without their prefixes.

config values are exposed as preprocessor definitions: unfortunately these have the potential to collide with any other tokens in source files. It's much better to create short friendly names for use in an API by either generating definitions based on the config (like ARMmbed/mbed-hal does), or by #if-ing on the preprocessor definitions to conditionally generate enum/class definitions with the preprocessor.

Hope that makes sense?

bridadan, 2016-02-08:
I agree that config values should not be exposed without their prefixes by default to prevent collisions. However I would argue that having a way to override the default prefix through the module.json/target.json would be far more user-friendly then hiding it in a module's source. It took me ~30 minutes to track down exactly where this was happening in the source, and I'm decently familiar with how the modules are organized. I can't imagine how an outside user would be able to find it!

Instead I think having this option in the JSON data would be much more intuitive, since I already by convention look there for config. A non-default prefix could even be displayed when doing `yotta config` and you could see from which module that prefix was applied.

closed 2017-03-28 by thegecko