ARMmbed/yotta #699: yotta docs should recommend inline namespaces

bremoran, 2016-02-05:
Yotta's namespacing recommendations could be improved. Instead of:

```namespace mymodule { 
    namespace current {
        using namespace ::mymodule::v1;
    }
}
```

Yottadocs should advocate ```inline``` namespaces in C++11 repositories instead:

```namespace mymodule { 
    namespace current {
        inline namespace ::mymodule::v1;
    }
}
```

This allows some improvements in how namespaces are handled:

e.g.:
```namespace lib
{
    inline namespace v2
    {
        template< class T >
        struct traits
        { /*...*/ };
   }
}
```

Then, we can do this:

```struct Mine
{};
namespace lib
{
template<>
struct traits< Mine >
{ /*...*/ };
}
```

With the ```using``` keyword, this template specialization is not possible.

cc @pan- @autopulated

autopulated, 2016-02-05:
:+1: This fixes a bunch of corner cases. You could also use namespace aliasing (`namespace current = ::mymodule::v1;`), which I think would have all the same advantages.

Both are C++11 features, so we should also note in the targets documentation that new targets should compile C++ code as C++11 (this is already the situation, but I don't think we state it explicitly).

pan-, 2016-02-05:
+1, 

@autopulated Yes, namespace aliasing can also solve the problem but it is more verbose ```lib::current::foo()``` is less intuitive than ```lib::foo()```

0xc0170, 2016-02-05:
+1 for these 2 new features, I noticed their use in boost , was planning to request this. For reference: https://svn.boost.org/trac/boost/wiki/BestPracticeHandbook