ARMmbed/yotta #701: Yotta fails to build with error "The command line is too long."

mazimkhan, 2016-02-09:
Yotta produces the linking command line with all objects paths. It becomes too long for Windows. Perhaps some other mechanism should be adopted while linking to avoid command lines longer than the limit. 

```
11:44:11 Started by upstream project "core-util-nightly-build/compiler=gcc,target=stm32f429i-disco-gcc" build number 8
11:44:11 originally caused by:
11:44:11  Started by upstream project "core-util-nightly-build" build number 8
11:44:11  originally caused by:
11:44:11   Started by user anonymous
11:44:11 [EnvInject] - Loading node environment variables.
11:44:11 Building remotely on cam_int_win_hw_3_builds (gcc windows armcc) in workspace c:\jslv3_1\ws\yb@10
11:44:11 [WS-CLEANUP] Deleting project workspace...
11:44:11 Cloning the remote Git repository
11:44:11 Cloning repository git@github.com:ARMmbed/core-util.git
11:44:11  > C:\Program Files\Git\cmd\git.exe init c:\jslv3_1\ws\yb@10 # timeout=10
11:44:12 Fetching upstream changes from git@github.com:ARMmbed/core-util.git
11:44:12  > C:\Program Files\Git\cmd\git.exe --version # timeout=10
11:44:12 using GIT_SSH to set credentials Jenkins master ssh keys
11:44:13  > C:\Program Files\Git\cmd\git.exe -c core.askpass=true fetch --tags --progress git@github.com:ARMmbed/core-util.git +refs/heads/*:refs/remotes/origin/*
11:44:15  > C:\Program Files\Git\cmd\git.exe config remote.origin.url git@github.com:ARMmbed/core-util.git # timeout=10
11:44:15  > C:\Program Files\Git\cmd\git.exe config --add remote.origin.fetch +refs/heads/*:refs/remotes/origin/* # timeout=10
11:44:15  > C:\Program Files\Git\cmd\git.exe config remote.origin.url git@github.com:ARMmbed/core-util.git # timeout=10
11:44:15 Fetching upstream changes from git@github.com:ARMmbed/core-util.git
11:44:15 using GIT_SSH to set credentials Jenkins master ssh keys
11:44:15  > C:\Program Files\Git\cmd\git.exe -c core.askpass=true fetch --tags --progress git@github.com:ARMmbed/core-util.git +refs/heads/*:refs/remotes/origin/*
11:44:17  > C:\Program Files\Git\cmd\git.exe rev-parse "refs/remotes/origin/master^{commit}" # timeout=10
11:44:17  > C:\Program Files\Git\cmd\git.exe rev-parse "refs/remotes/origin/origin/master^{commit}" # timeout=10
11:44:17 Checking out Revision ba6b6ad7a6be2d1a8b67e9208f2874202f164ea3 (refs/remotes/origin/master)
11:44:17  > C:\Program Files\Git\cmd\git.exe config core.sparsecheckout # timeout=10
11:44:17  > C:\Program Files\Git\cmd\git.exe checkout -f ba6b6ad7a6be2d1a8b67e9208f2874202f164ea3
11:44:17  > C:\Program Files\Git\cmd\git.exe rev-list e20c8c58fc8fba78ec581c84ff1f6e1b6a4a5675 # timeout=10
11:44:18 First time build. Skipping changelog.
11:44:18 Cloning the remote Git repository
11:44:18 Cloning repository git@github.com:ARMmbed/update-tools.git
11:44:18  > C:\Program Files\Git\cmd\git.exe init c:\jslv3_1\ws\yb@10\update-tools # timeout=10
11:44:18 Fetching upstream changes from git@github.com:ARMmbed/update-tools.git
11:44:18  > C:\Program Files\Git\cmd\git.exe --version # timeout=10
11:44:18 using GIT_SSH to set credentials Jenkins master ssh keys
11:44:18  > C:\Program Files\Git\cmd\git.exe -c core.askpass=true fetch --tags --progress git@github.com:ARMmbed/update-tools.git +refs/heads/*:refs/remotes/origin/*
11:44:34  > C:\Program Files\Git\cmd\git.exe config remote.origin.url git@github.com:ARMmbed/update-tools.git # timeout=10
11:44:34  > C:\Program Files\Git\cmd\git.exe config --add remote.origin.fetch +refs/heads/*:refs/remotes/origin/* # timeout=10
11:44:34  > C:\Program Files\Git\cmd\git.exe config remote.origin.url git@github.com:ARMmbed/update-tools.git # timeout=10
11:44:34 Fetching upstream changes from git@github.com:ARMmbed/update-tools.git
11:44:34 using GIT_SSH to set credentials Jenkins master ssh keys
11:44:34  > C:\Program Files\Git\cmd\git.exe -c core.askpass=true fetch --tags --progress git@github.com:ARMmbed/update-tools.git +refs/heads/*:refs/remotes/origin/*
11:44:36  > C:\Program Files\Git\cmd\git.exe rev-parse "refs/remotes/origin/master^{commit}" # timeout=10
11:44:36  > C:\Program Files\Git\cmd\git.exe rev-parse "refs/remotes/origin/origin/master^{commit}" # timeout=10
11:44:36 Checking out Revision d919f6b62fd29d96b8e5e6b4df97368cb25deeb5 (refs/remotes/origin/master)
11:44:36  > C:\Program Files\Git\cmd\git.exe config core.sparsecheckout # timeout=10
11:44:36  > C:\Program Files\Git\cmd\git.exe checkout -f d919f6b62fd29d96b8e5e6b4df97368cb25deeb5
11:44:36  > C:\Program Files\Git\cmd\git.exe rev-list d919f6b62fd29d96b8e5e6b4df97368cb25deeb5 # timeout=10
11:44:36 Next nodes: []
11:44:36 [yb@10] $ sh -xe C:\Windows\TEMP\hudson8692351234540348915.sh
11:44:36 + rm -rf venv
11:44:36 + virtualenv venv
11:44:43 New python executable in c:\jslv3_1\ws\yb@10\venv\Scripts\python.exe
11:45:40 Installing setuptools, pip, wheel...done.
11:45:40 + source venv/Scripts/activate
11:45:40 ++ deactivate nondestructive
11:45:40 ++ unset -f pydoc
11:45:40 ++ '[' -z '' ']'
11:45:40 ++ '[' -z '' ']'
11:45:40 ++ '[' -n /usr/bin/sh ']'
11:45:40 ++ hash -r
11:45:40 ++ '[' -z '' ']'
11:45:40 ++ unset VIRTUAL_ENV
11:45:40 ++ '[' '!' nondestructive = nondestructive ']'
11:45:40 +++ '[' msys == cygwin ']'
11:45:40 +++ echo /c/jslv3_1/ws/yb@10/venv
11:45:40 ++ VIRTUAL_ENV=/c/jslv3_1/ws/yb@10/venv
11:45:40 ++ export VIRTUAL_ENV
[... 272 lines cut ...]
11:55:01 [142/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_cec.c.o
11:55:01 [143/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_i2c.c.o
11:55:01 [144/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_pwr.c.o
11:55:01 [145/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_wwdg.c.o
11:55:02 [146/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_sai_ex.c.o
11:55:02 [147/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_rng.c.o
11:55:02 [148/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_pcd_ex.c.o
11:55:02 [149/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_irda.c.o
11:55:02 [150/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_i2s_ex.c.o
11:55:03 [151/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_nand.c.o
11:55:03 [152/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_rcc.c.o
11:55:03 [153/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_cryp.c.o
11:55:03 [154/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_sram.c.o
11:55:03 [155/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_hcd.c.o
11:55:03 [156/176] Building C object ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_dac.c.o
11:55:04 [157/176] Linking CXX static library ym/mbed-hal-st-stm32cubef4/source/mbed-hal-st-stm32cubef4.a
[... cut ...] 3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_cryp.c.o ym/mbed-hal-st-stm32cubef4/source/CMakeFiles/mbed-hal-st-stm32cubef4.dir/c_/jslv3_1/ws/yb@10/yotta_modules/mbed-hal-st-stm32cubef4/source/stm32f4xx_hal_adc.c.o && cd ."
11:55:04 The command line is too long.

11:55:04 ninja: build stopped: subcommand failed.
11:55:04 error: command ['ninja'] failed
11:55:04 Build step 'Execute shell' marked build as failure
11:55:05 Archiving artifacts
11:55:06 Finished: FAILURE
```

autopulated, 2016-02-09:
I believe that a linker response file (a file which lists the objects to be linked, instead of listing them on the command line) should be being used for this step (@filename on the command line).

Could you confirm if the build works with an armcc based target, and also what versions of ninja and cmake you have?

mazimkhan, 2016-02-09:
It is mostly failing for gcc. Haven't seen this error for armcc. Although for armcc we have license server issues. So it is not being fully tested.

cmake version 3.2.1
ninja version 1.5.3

autopulated, 2016-02-09:
Please could you try updating cmake and ninja to >=3.4 and >=1.6 respectively, and confirm if this is still an issue for a clean build?

mazimkhan, 2016-02-09:
Upgraded cmake and ninja. Still having the same issue

http://e108747.cambridge.arm.com:8080/job/yb/684/console
http://e108747.cambridge.arm.com:8080/job/yb/690/console
http://e108747.cambridge.arm.com:8080/job/yb/687/console

Plus after upgrade following exception is raised by Yotta
http://e108747.cambridge.arm.com:8080/job/yb/681/console
http://e108747.cambridge.arm.com:8080/job/yb/682/console

mazimkhan, 2016-04-05:
This issue is still at large
http://e108747.cambridge.arm.com:8080/job/yb/7149/console
http://e108747.cambridge.arm.com:8080/job/yb/7202/console
http://e108747.cambridge.arm.com:8080/job/yb/7198/console
... and many more

This at this time is only issue affecting CI jobs. Can you please look at it.

autopulated, 2016-04-05:
For some reason the linker response file isn't being used, when it should be.

Unfortunately I haven't been able to access the build logs you reference.

It's worth checking that CMAKE_C_RESPONSE_FILE_LINK_FLAG and CMAKE_CXX_RESPONSE_FILE_LINK_FLAG are being set correctly in the target you're using (or one of the targets it inherits from). (It should be set to "@", I think)