ARMmbed/yotta #702: CMake Deprecation Warning

PrzemekWirkus, 2016-02-10:
# Description
Got this sort of warnings after updating CMake and ninja:
```
The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
```

## Software versions:
Platform: Windows
* cmake version 3.5.0-rc1
* ninja: 1.6.0
* yotta: 0.13.1

## Full warning listing
```
[master] C:\Work\mbed-drivers> yt build
```
```
warning: The build target has not been set, so the default (frdm-k64f-gcc,*) is being used. You can use `yotta target <targetname>` to set the build target. See http://yottadocs.mbed.com/tutorial/targets.html for more information on using targets.
info: generate for target: frdm-k64f-gcc 2.0.0 at c:\Work\mbed-drivers\yotta_targets\frdm-k64f-gcc
CMake Deprecation Warning at C:/Program Files (x86)/CMake/share/cmake-3.5/Modules/CMakeForceCompiler.cmake:73 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Work/mbed-drivers/yotta_targets/mbed-gcc/CMake/toolchain.cmake:78 (cmake_force_c_compiler)
  toolchain.cmake:8 (include)
  C:/Program Files (x86)/CMake/share/cmake-3.5/Modules/CMakeDetermineSystem.cmake:98 (include)
  CMakeLists.txt:148 (project)

CMake Deprecation Warning at C:/Program Files (x86)/CMake/share/cmake-3.5/Modules/CMakeForceCompiler.cmake:87 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Work/mbed-drivers/yotta_targets/mbed-gcc/CMake/toolchain.cmake:79 (cmake_force_cxx_compiler)
  toolchain.cmake:8 (include)
  C:/Program Files (x86)/CMake/share/cmake-3.5/Modules/CMakeDetermineSystem.cmake:98 (include)
  CMakeLists.txt:148 (project)

GCC version is: 4.8.4
-- The ASM compiler identification is GNU
-- Found assembler: C:/Work/toolchains/gcc_arm_4_8/4_8_2014q3/bin/arm-none-eabi-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Work/mbed-drivers/build/frdm-k64f-gcc
[63/168] Building CXX object ym/minar-platform-mbed/source/CMakeFiles/minar-platform-mbed.dir/C_/Work/mbed-drivers/yotta_modules/minar-platform-mbed/source/mbed_platform.cpp.o
C:/Work/mbed-drivers/yotta_modules/minar-platform-mbed/source/mbed_platform.cpp:39:32: warning: 'last_wakeup' defined but not used [-Wunused-variable]
 static minar::platform::tick_t last_wakeup = 0;
                                ^
C:/Work/mbed-drivers/yotta_modules/minar-platform-mbed/source/mbed_platform.cpp:40:32: warning: 'last_gotosleep' defined but not used [-Wunused-variable]
 static minar::platform::tick_t last_gotosleep = 0;
                                ^
[168/168] Linking CXX executable test/mbed-drivers-test-serial_interrupt
```

## Other stuff
```
$ yt target
frdm-k64f-gcc 2.0.0
kinetis-k64-gcc 2.0.0
mbed-gcc 1.1.0
```
```
$ yt ls
mbed-drivers 0.12.1
|_ mbed-hal 1.2.2
| \_ mbed-hal-freescale 1.0.0 yotta_modules\mbed-hal-freescale
|   \_ mbed-hal-ksdk-mcu 1.0.8 yotta_modules\mbed-hal-ksdk-mcu
|     |_ uvisor-lib 1.0.12 yotta_modules\uvisor-lib
|     \_ mbed-hal-k64f 1.0.1 yotta_modules\mbed-hal-k64f
|       \_ mbed-hal-frdm-k64f 1.0.0 yotta_modules\mbed-hal-frdm-k64f
|_ cmsis-core 1.1.2
| \_ cmsis-core-freescale 1.0.0 yotta_modules\cmsis-core-freescale
|   \_ cmsis-core-k64f 1.0.0 yotta_modules\cmsis-core-k64f
|_ ualloc 1.0.3
| \_ dlmalloc 1.0.0 yotta_modules\dlmalloc
|_ minar 1.0.4
| \_ minar-platform 1.0.0 yotta_modules\minar-platform
|   \_ minar-platform-mbed 1.1.1 yotta_modules\minar-platform-mbed
|_ core-util 1.2.0
\_ compiler-polyfill 1.2.1
```

autopulated, 2016-02-10:
Thanks for reporting this!

Possibly for arm-none-eabi-gcc we can avoid using this now, but I think for armcc we will still need to do something similar.

autopulated, 2016-03-18:
Not deprecated for now, but support has been added upstream and in mbed-gcc and mbed-armcc so when it is deprecated it won't be a problem.

closed 2016-03-18 by autopulated