ARMmbed/yotta #707: Yotta build causes exception "WindowsError: [Error 183] Cannot create a file when that file already exists"

mazimkhan, 2016-02-10:
Yotta build causes below exception. cmake version 3.4, ninja version 1.6

09:57:04 Started by upstream project "bt" build number 5
09:57:04 originally caused by:
09:57:04  Started by upstream project "mbed-drivers-nightly-build-and-test2/compiler=armcc,target=frdm-k64f-armcc" build number 8
09:57:04  originally caused by:
09:57:04   Started by upstream project "mbed-drivers-nightly-build-and-test2" build number 8
09:57:04   originally caused by:
09:57:04    Started by user mazimkhan
09:57:04 [EnvInject] - Loading node environment variables.
09:57:04 Building remotely on cam_int_win_hw_3_builds (gcc windows armcc) in workspace c:\jslv3_1\ws\yb@3
09:57:04 [WS-CLEANUP] Deleting project workspace...
09:57:11 [WS-CLEANUP] Done
09:57:11 Cloning the remote Git repository
09:57:11 Cloning repository git@github.com:ARMmbed/mbed-drivers.git
09:57:11  > C:\Program Files\Git\cmd\git.exe init c:\jslv3_1\ws\yb@3 # timeout=10
09:57:11 Fetching upstream changes from git@github.com:ARMmbed/mbed-drivers.git
09:57:11  > C:\Program Files\Git\cmd\git.exe --version # timeout=10
09:57:11 using GIT_SSH to set credentials Jenkins master ssh keys
09:57:11  > C:\Program Files\Git\cmd\git.exe -c core.askpass=true fetch --tags --progress git@github.com:ARMmbed/mbed-drivers.git +refs/heads/*:refs/remotes/origin/*
09:59:30  > C:\Program Files\Git\cmd\git.exe config remote.origin.url git@github.com:ARMmbed/mbed-drivers.git # timeout=10
09:59:30  > C:\Program Files\Git\cmd\git.exe config --add remote.origin.fetch +refs/heads/*:refs/remotes/origin/* # timeout=10
09:59:30  > C:\Program Files\Git\cmd\git.exe config remote.origin.url git@github.com:ARMmbed/mbed-drivers.git # timeout=10
09:59:30 Fetching upstream changes from git@github.com:ARMmbed/mbed-drivers.git
09:59:30 using GIT_SSH to set credentials Jenkins master ssh keys
09:59:30  > C:\Program Files\Git\cmd\git.exe -c core.askpass=true fetch --tags --progress git@github.com:ARMmbed/mbed-drivers.git +refs/heads/*:refs/remotes/origin/*
09:59:31  > C:\Program Files\Git\cmd\git.exe rev-parse "refs/remotes/origin/master^{commit}" # timeout=10
09:59:31  > C:\Program Files\Git\cmd\git.exe rev-parse "refs/remotes/origin/origin/master^{commit}" # timeout=10
09:59:31 Checking out Revision bfdff4582c170f72fee86ed2ab5fbe3b602a2c0b (refs/remotes/origin/master)
09:59:31  > C:\Program Files\Git\cmd\git.exe config core.sparsecheckout # timeout=10
09:59:31  > C:\Program Files\Git\cmd\git.exe checkout -f bfdff4582c170f72fee86ed2ab5fbe3b602a2c0b
09:59:32  > C:\Program Files\Git\cmd\git.exe rev-list bfdff4582c170f72fee86ed2ab5fbe3b602a2c0b # timeout=10
09:59:32 Cloning the remote Git repository
09:59:32 Cloning repository git@github.com:ARMmbed/update-tools.git
09:59:32  > C:\Program Files\Git\cmd\git.exe init c:\jslv3_1\ws\yb@3\update-tools # timeout=10
09:59:33 Fetching upstream changes from git@github.com:ARMmbed/update-tools.git
09:59:33  > C:\Program Files\Git\cmd\git.exe --version # timeout=10
09:59:33 using GIT_SSH to set credentials Jenkins master ssh keys
09:59:33  > C:\Program Files\Git\cmd\git.exe -c core.askpass=true fetch --tags --progress git@github.com:ARMmbed/update-tools.git +refs/heads/*:refs/remotes/origin/*
09:59:38  > C:\Program Files\Git\cmd\git.exe config remote.origin.url git@github.com:ARMmbed/update-tools.git # timeout=10
09:59:39  > C:\Program Files\Git\cmd\git.exe config --add remote.origin.fetch +refs/heads/*:refs/remotes/origin/* # timeout=10
09:59:39  > C:\Program Files\Git\cmd\git.exe config remote.origin.url git@github.com:ARMmbed/update-tools.git # timeout=10
09:59:39 Fetching upstream changes from git@github.com:ARMmbed/update-tools.git
09:59:39 using GIT_SSH to set credentials Jenkins master ssh keys
09:59:39  > C:\Program Files\Git\cmd\git.exe -c core.askpass=true fetch --tags --progress git@github.com:ARMmbed/update-tools.git +refs/heads/*:refs/remotes/origin/*
09:59:40  > C:\Program Files\Git\cmd\git.exe rev-parse "refs/remotes/origin/master^{commit}" # timeout=10
09:59:40  > C:\Program Files\Git\cmd\git.exe rev-parse "refs/remotes/origin/origin/master^{commit}" # timeout=10
09:59:40 Checking out Revision 352babe8f22f89a18918653fcf099420e54e4205 (refs/remotes/origin/master)
09:59:40  > C:\Program Files\Git\cmd\git.exe config core.sparsecheckout # timeout=10
09:59:40  > C:\Program Files\Git\cmd\git.exe checkout -f 352babe8f22f89a18918653fcf099420e54e4205
09:59:40  > C:\Program Files\Git\cmd\git.exe rev-list 352babe8f22f89a18918653fcf099420e54e4205 # timeout=10
09:59:40 Next nodes: []
09:59:41 [yb@3] $ sh -xe C:\Windows\TEMP\hudson3474165582156421340.sh
09:59:41 + rm -rf venv
09:59:41 + virtualenv venv
09:59:43 New python executable in c:\jslv3_1\ws\yb@3\venv\Scripts\python.exe
10:00:21 Installing setuptools, pip, wheel...done.
10:00:21 + source venv/Scripts/activate
10:00:21 ++ deactivate nondestructive
10:00:21 ++ unset -f pydoc
10:00:21 ++ '[' -z '' ']'
10:00:21 ++ '[' -z '' ']'
10:00:21 ++ '[' -n /usr/bin/sh ']'
10:00:21 ++ hash -r
10:00:21 ++ '[' -z '' ']'
10:00:21 ++ unset VIRTUAL_ENV
10:00:21 ++ '[' '!' nondestructive = nondestructive ']'
10:00:21 +++ '[' msys == cygwin ']'
10:00:21 +++ echo /c/jslv3_1/ws/yb@3/venv
10:00:21 ++ VIRTUAL_ENV=/c/jslv3_1/ws/yb@3/venv
[... 134 lines cut ...]
10:04:09     name, version_required, install_into, v, type=type, inherit_shrinkwrap = inherit_shrinkwrap
10:04:09   File "c:\jslv3_1\ws\yb@3\venv\lib\site-packages\yotta\lib\access.py", line 318, in _satisfyVersionByInstallingVersion
10:04:09     version.unpackInto(working_directory)
10:04:09   File "c:\jslv3_1\ws\yb@3\venv\lib\site-packages\yotta\lib\registry_access.py", line 447, in unpackInto
10:04:09     _getTarball(self.url, directory, self.sha256)
10:04:09   File "c:\jslv3_1\ws\yb@3\venv\lib\site-packages\yotta\lib\registry_access.py", line 100, in wrapped
10:04:09     return fn(*args, **kwargs)
10:04:09   File "c:\jslv3_1\ws\yb@3\venv\lib\site-packages\yotta\lib\registry_access.py", line 185, in wrapped
10:04:09     return fn(*args, **kwargs)
10:04:09   File "c:\jslv3_1\ws\yb@3\venv\lib\site-packages\yotta\lib\registry_access.py", line 168, in wrapped
10:04:09     return fn(*args, **kwargs)
10:04:09   File "c:\jslv3_1\ws\yb@3\venv\lib\site-packages\yotta\lib\registry_access.py", line 149, in wrapped
10:04:09     return fn(*args, **kwargs)
10:04:09   File "c:\jslv3_1\ws\yb@3\venv\lib\site-packages\yotta\lib\registry_access.py", line 296, in _getTarball
10:04:09     origin_info = {'url':url}
10:04:09   File "c:\jslv3_1\ws\yb@3\venv\lib\site-packages\yotta\lib\access_common.py", line 124, in wrapped
10:04:09     r = fn(*args, **kwargs)
10:04:09   File "c:\jslv3_1\ws\yb@3\venv\lib\site-packages\yotta\lib\access_common.py", line 288, in unpackTarballStream
10:04:09     new_cache_key = downloadToCache(stream, hash, cache_key, origin_info)
10:04:09   File "c:\jslv3_1\ws\yb@3\venv\lib\site-packages\yotta\lib\access_common.py", line 261, in downloadToCache
10:04:09     os.rename(download_fname, cache_as)
10:04:09 WindowsError: [Error 183] Cannot create a file when that file already exists
10:04:10 Build step 'Execute shell' marked build as failure
10:04:10 Archiving artifacts
10:04:10 Finished: FAILURE

autopulated, 2016-02-10:
looks like a bug in the caching, you can probably workaround it by removing ~/.yotta/cache

mazimkhan, 2016-02-10:
All Jenkins workspaces are created from scratch. there should not be a ~/.yotta/cache file. Are you suggesting it happens due to stale ~/.yotta/cache file.

autopulated, 2016-02-10:
The point where the error occurs is where a downloaded module is being put into the cache, and the error implies the same file was already present (which is a case that should be handled), hence my suggestion, but maybe the file was put there by a parallel build, for example.

mazimkhan, 2016-02-10:
Oh I think I overlooked the path. So the .yotta dir is in home directory. In automated builds I don't want to delete this file from home. As I don't know what other builds would be doing with it. Its a shared file created by yotta and yotta should be able to manage its states and contingencies.

Can we have a no cache option in yotta or we can specify the cache in the workspace that is local to the build. Perhaps later would defy the purpose of it.

autopulated, 2016-02-10:
You normally want the cache, especially if you are building similar things many times, because it will reduce duplicate downloads. Configure the cache size with `"maxCachedModules":<number>` in your `~/.yotta/config.json` file: you could set this to zero to temporarily disable it.

mazimkhan, 2016-02-11:
Yes the feature is good. But it does not support parallel instances of yotta. Do you think you would be fixing this issue near future. By that time we can disable the cache size.

autopulated, 2016-02-11:
The cache *should* support parallel instances of yotta, that's why this is a bug :)

mazimkhan, 2016-02-15:
Had updated ~/.yotta/config.json with ```"maxCachedModules": 0```
```
{
  "keys": {
    "private": xyz,
    "public": xyz,
  },
  "uuid": xyz,
  "mbed": {
    "authtoken": xyz
  },
  "maxCachedModules": 0
}
```

Still I am getting the same exception:

http://e108747.cambridge.arm.com:8080/job/ble-examples-nightly-build/label=gcc,target=nrf51dk-gcc/1/console

autopulated, 2016-02-15:
OK thanks for trying that, I will look at this soon.

autopulated, 2016-02-18:
Should be fixed by #713.

closed 2016-03-18 by autopulated