ARMmbed/yotta #708: yotta command line tool requires ARM mbed account

kilogram, 2016-02-12:
The yotta tool requires a registered, verified account with ARM mbed in order to function correctly. 

While ARM mbed accounts are freely available to the public, this is atypical behaviour in open source projects, especially when the functions it is performing is cloning/downloading drivers and libraries which should also be publicly available.

yotta may have the option of using an account, since it appears that there is an attempt to build a community for mbed projects; however, it should not be required.

hugovincent, 2016-02-12:
Hi @kilogram we're aware of this issue -- right now we need this for ARM-internal process reasons but we're working to find a way of removing the need for the login.

closed 2016-02-12 by hugovincent

thegecko, 2017-04-24:
We have removed the need to log in to yotta for read-only access.