ARMmbed/yotta #712: Fatal Exception, yotta=0.13.1 (WindowsError: [Error 183] Cannot create a file when that file already exists)

PrzemekWirkus, 2016-02-17:
# Description
When building in our CI repo ```https://github.com/PrzemekWirkus/utest/tree/devel_transport``` I got below error:
```
$ cd utest
$ yotta target frdm-k64f-armcc
```
```python
Fatal Exception, yotta=0.13.1
Traceback (most recent call last):
  File "c:\python27\Lib\runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\jslv3_1\ws\ybt@2\venv\Scripts\yotta.exe\__main__.py", line 9, in <module>
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\main.py", line 206, in main
    status = args.command(args, following_args)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\build.py", line 46, in execCommand
    status = installAndBuild(args, following_args)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\build.py", line 98, in installAndBuild
    install_status = install.execCommand(args, [])
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\install.py", line 62, in execCommand
    return installDeps(args, c)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\install.py", line 119, in installDeps
    test = {'own':'toplevel', 'all':True, 'none':False}[args.install_test_deps]
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\component.py", line 651, in satisfyDependenciesRecursive
    test = test
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\component.py", line 415, in __getDependenciesRecursiveWithProvider
    test = test
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\component.py", line 315, in __getDependenciesWithProvider
    satisfyDep, specs
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\component.py", line 291, in satisfyDep
    self
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\component.py", line 636, in provider
    inherit_shrinkwrap = dep_of.getShrinkwrap()
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\access.py", line 307, in satisfyVersionByInstalling
    name, version_required, install_into, v, type=type, inherit_shrinkwrap = inherit_shrinkwrap
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\access.py", line 318, in _satisfyVersionByInstallingVersion
    version.unpackInto(working_directory)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 447, in unpackInto
    _getTarball(self.url, directory, self.sha256)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 100, in wrapped
    return fn(*args, **kwargs)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 185, in wrapped
    return fn(*args, **kwargs)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 168, in wrapped
    return fn(*args, **kwargs)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 149, in wrapped
    return fn(*args, **kwargs)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 296, in _getTarball
    origin_info = {'url':url}
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\access_common.py", line 124, in wrapped
    r = fn(*args, **kwargs)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\access_common.py", line 288, in unpackTarballStream
    new_cache_key = downloadToCache(stream, hash, cache_key, origin_info)
  File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\access_common.py", line 261, in downloadToCache
    os.rename(download_fname, cache_as)
WindowsError: [Error 183] Cannot create a file when that file already exists
```

### Direct log from our (Windows 2012 Server) CI node:
```
16:05:37 + yotta target frdm-k64f-armcc
16:05:37 info: get versions for frdm-k64f-armcc
16:05:38 info: download frdm-k64f-armcc@2.0.0 from the public module registry
16:05:38 info: get versions for kinetis-k64-armcc
16:05:39 info: download kinetis-k64-armcc@2.0.0 from the public module registry
16:05:39 info: get versions for mbed-armcc
16:05:39 info: download mbed-armcc@1.0.1 from the public module registry
16:05:40 + yotta build
16:05:40 info: get versions for minar
16:05:41 info: download minar@1.0.4 from the public module registry
16:05:41 info: get versions for core-util
16:05:42 info: download core-util@1.3.0 from the public module registry
16:05:42 info: get versions for compiler-polyfill
16:05:42 info: download compiler-polyfill@1.2.1 from the public module registry
16:05:44 info: get versions for mbed-drivers
16:05:45 info: download mbed-drivers@0.12.1 from the public module registry
16:05:46 info: get versions for greentea-client
16:05:47 info: download greentea-client@0.1.3 from the public module registry
16:05:49 Fatal Exception, yotta=0.13.1
16:05:49 Traceback (most recent call last):
16:05:49   File "c:\python27\Lib\runpy.py", line 162, in _run_module_as_main
16:05:49     "__main__", fname, loader, pkg_name)
16:05:49   File "c:\python27\Lib\runpy.py", line 72, in _run_code
16:05:49     exec code in run_globals
16:05:49   File "C:\jslv3_1\ws\ybt@2\venv\Scripts\yotta.exe\__main__.py", line 9, in <module>
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\main.py", line 206, in main
16:05:49     status = args.command(args, following_args)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\build.py", line 46, in execCommand
16:05:49     status = installAndBuild(args, following_args)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\build.py", line 98, in installAndBuild
16:05:49     install_status = install.execCommand(args, [])
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\install.py", line 62, in execCommand
16:05:49     return installDeps(args, c)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\install.py", line 119, in installDeps
16:05:49     test = {'own':'toplevel', 'all':True, 'none':False}[args.install_test_deps]
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\component.py", line 651, in satisfyDependenciesRecursive
16:05:49     test = test
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\component.py", line 415, in __getDependenciesRecursiveWithProvider
16:05:49     test = test
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\component.py", line 315, in __getDependenciesWithProvider
16:05:49     satisfyDep, specs
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\component.py", line 291, in satisfyDep
16:05:49     self
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\component.py", line 636, in provider
16:05:49     inherit_shrinkwrap = dep_of.getShrinkwrap()
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\access.py", line 307, in satisfyVersionByInstalling
16:05:49     name, version_required, install_into, v, type=type, inherit_shrinkwrap = inherit_shrinkwrap
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\access.py", line 318, in _satisfyVersionByInstallingVersion
16:05:49     version.unpackInto(working_directory)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 447, in unpackInto
16:05:49     _getTarball(self.url, directory, self.sha256)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 100, in wrapped
16:05:49     return fn(*args, **kwargs)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 185, in wrapped
16:05:49     return fn(*args, **kwargs)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 168, in wrapped
16:05:49     return fn(*args, **kwargs)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 149, in wrapped
16:05:49     return fn(*args, **kwargs)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\registry_access.py", line 296, in _getTarball
16:05:49     origin_info = {'url':url}
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\access_common.py", line 124, in wrapped
16:05:49     r = fn(*args, **kwargs)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\access_common.py", line 288, in unpackTarballStream
16:05:49     new_cache_key = downloadToCache(stream, hash, cache_key, origin_info)
16:05:49   File "c:\jslv3_1\ws\ybt@2\venv\lib\site-packages\yotta\lib\access_common.py", line 261, in downloadToCache
16:05:49     os.rename(download_fname, cache_as)
16:05:49 WindowsError: [Error 183] Cannot create a file when that file already exists
```

autopulated, 2016-02-18:
This should be fixed in #713, please try and let me know!

PrzemekWirkus, 2016-02-18:
James,
I can't reproduce on my local machine.

This particular is running on CI system so I will report it if this will reoccur.
Our Ci is installing in virtualenv ```yotta``` from PyPi so for us this fix comes when you release to PyPi! :)

autopulated, 2016-02-18:
OK :+1:

PR #713 (2016-02-18): Various Cache-related Improvements
 * improve handling of possible race conditions, especially on windows (fixes #712)
 * increase default cache size
 * ensure that setting the cache size to zero actually disables it
closed 2016-02-18 by autopulated