ARMmbed/yotta #716: Library source code directory can't be subdir or root dir

aamcampbell, 2016-02-19:
The documentation for the module.json file clearly states that a subdirectory may be used for the "lib" property of a library module:

> For example, to build a library out of the contents of ./some/subdirectory instead of the default source directory, use:
> 
>   "lib": "./some/subdirectory"
(http://yottadocs.mbed.com/reference/module.html)

However, this does not work for me. I have a "source" folder which is empty except for a "subsrc" directory, which contains source code. If I set `"lib": "./source"`, my source files are compiled correctly. If I set `"lib": "./source/subsrc"`, yotta finds no sources and generates an empty placeholder library instead.

The same thing happens when I set `"lib": "./"` (no sources are found). The behavior of using the root directory as the source dir is not clearly documented, but it seems like it ought to be allowed.

autopulated, 2016-02-20:
Thanks for reporting!

The subdirectory should work, putting source files directly in the root directory is probably not a good idea, but I think it should be supported.

aamcampbell, 2016-02-22:
Yes, it works fine with `./anyfolder` or just `anyfolder`, but not with `./folder1/folder2`. I haven't tried `./source/folder2`.

The reason I might use the root directory for sources is to maintain compatibility with other build systems. A custom cmake is probably a better idea, but it would be nice to choose.